dgibson/dtc #1: Some of the new functions are missing from the version.lds file.

kenaaker, 2015-11-19:
I was working the a 1.4.1 version built on the OpenSuSE factory build setup and found that the fdt_first_subnode() function wasn't usable from the shared library. I found a few other functions that were listed in the .h file were also marked as locals in the shared library. The list I patched into version.lds was

		fdt_first_subnode;
		fdt_next_subnode;
		fdt_address_cells;
		fdt_size_cells;
		fdt_stringlist_contains;
		fdt_resize;

I don't know if any others are missing or if there are extras.

dgibson, 2015-11-20:
On Thu, Nov 19, 2015 at 03:05:04PM -0800, Ken Aaker wrote:
> I was working the a 1.4.1 version built on the OpenSuSE factory build setup and found that the fdt_first_subnode() function wasn't usable from the shared library. I found a few other functions that [...]

These two were already added by commit a4b093f7366fdb429ca1781144d3985fa50d0fbb

> 		fdt_address_cells;
> 		fdt_size_cells;
> 		fdt_stringlist_contains;
> 		fdt_resize;

I have a draft patch adding these.  Can I get an email for you to put
int a Reported-by line?

-- 
David Gibson			| I'll have my music baroque, and my code
david AT gibson.dropbear.id.au	| minimalist, thank you.  NOT _the_ _other_
				| _way_ _around_!
http://www.ozlabs.org/~dgibson

kenaaker, 2015-11-20:
Hello David,

You can use the e-mail that I sent this from.

I'm trying to update a board running a 2.6.21 kernel on an Atmel armv5
SOC. So, I'm having a bit of trouble getting a proper device tree set
up. I have started to write something that will display the dtb that I
get so I can see whether I've managed to get the mdio and mii bus
definitions set up properly. That was where I ran into the missing
global external for fdt_first_subnode.

Thank you for supporting the dtc.

Ken Aaker

closed 2016-05-31 by dgibson