dgibson/dtc #5: Invalid format string in a call to die()

fishilico, 2017-02-24:
Hello,
In order to detect printf format issues, I am compiling Linux kernel with options like -Wmissing-format-attribute -Wformat=2. This lead me to add  `__attribute__ ((format(printf, 1, 2)))` to `lexical_error()` in dtc-lexer.l and `die()` in util.h.
Doing so triggered a compiler warning on the following code in dtc.c:

		case 'a':
			alignsize = strtol(optarg, NULL, 0);
			if (!is_power_of_2(alignsize))
				die("Invalid argument \"%d\" to -a option\n",
				    optarg);
			break;

As variable `optarg` is a string, it should not be formatted with %d but with %s. As I have not found the mailing list archive of devicetree-compiler@vger.kernel.org (http://vger.kernel.org/vger-lists.html#devicetree-compiler references a link to http://dir.gmane.org/gmane.comp.devicetree.compiler which is empty), I do not know whether this bug has already been reported on the mailing list. This is why I am reporting it on Github.

Thanks

dgibson, 2017-02-27:
Hi,  the github issue tracker doesn't get used a whole lot.

I've made draft patches to fix this bug and add the printf warnings.  Can you contact me by email to let me know your email so I can credit you in the commit messages.

dgibson, 2017-02-27:
Fix has been pushed to master.

closed 2017-02-27 by dgibson