dgibson/dtc #8: python check

lordelectrode, 2017-08-16:
https://github.com/dgibson/dtc/blob/master/Makefile#L132
Only python2 and python3 exist on archlinux (which i presume is as close to upstream as possible)

dgibson, 2017-08-16:
I really have no idea what you're trying to say here.

lordelectrode, 2017-08-16:
`$(PKG_CONFIG) --cflags python` will always fail regardless of either/both python versions being installed

lordelectrode, 2017-08-16:
`if (layer > 0x01)` seems to work

dgibson, 2017-08-16:
So, changing it to $(PKG_CONFIG) --cflags python2 might work, since I believe the Python code is built for Python2, not Python3.  I'd have to check if that still works on more common distros than archlinux, though.

lordelectrode, 2017-08-16:
https://www.python.org/dev/peps/pep-0394/

dgibson, 2017-09-27:
I just pushed a commit to fix this to master.

Sorry it's taken so long, I've been a bit swamped.  Mind you, making your report in clear sentences rather than cryptic URL references wouldn't have hurt.

closed 2017-09-27 by dgibson