dgibson/dtc #9: failing tests on arm32

vagrantc, 2017-10-05:
82 tests fail on 32-bit arm systems on Debian's build infrastructure for both armel and armhf, both 32-bit arm architectures (arm64/aarch64 tests fine). 1.4.5-2 in Debian includes a patch from upstream 497432fd2131967f349e69dc5d259072151cc4b4: checks: Use proper format modifier for size_t (without it the package failed to build much earlier).

See the build logs for a little more information:

https://buildd.debian.org/status/fetch.php?pkg=device-tree-compiler&arch=armel&ver=1.4.5-2&stamp=1506657642&raw=0

https://buildd.debian.org/status/fetch.php?pkg=device-tree-compiler&arch=armhf&ver=1.4.5-2&stamp=1506657442&raw=0

Thanks!

dgibson, 2017-10-06:
Looks like this isn't new, every version I could find seems to have the same problem.

It looks like many - maybe all - of the failures are because we're not correctly building trees.S, where we (ab)use the assembler to make us a sample dtb to test against.  Specifically generating the memory reserve entries there uses 64-bit constant arithmetic, which it seems the arm32 as doesn't support.

What's not clear to me is why the arm32 as doesn't support it, but, for example, the i386 as does.

dgibson, 2017-10-06:
I just committed code to dtc master to stop it relying on 64-bit arithmetic in the assembler.  This should fix the problem.

dgibson, 2017-10-06:
Note, by the way, that the problem was in the testsuite - AFAICT dtc and libfdt themselves were building fine.

closed 2017-10-06 by dgibson