dgibson/dtc #15: test failures in pylibfdt on armv7

sharkcz, 2018-08-22:
I've enabled "make check" in our CI for dtc and it shows 3 errors when run on armv7 (Fedora 28). All checks pass on aarch64, ppc64, ppc64le, s390x and x86_64.
````
...
Check that exceptions can be masked by QUIET_ALL ... ok
testReserveMap (__main__.PyLibfdtBasicTests)
Test that we can access the memory reserve map ... FAIL
testSetName (__main__.PyLibfdtBasicTests)
Test that we can update a node name ... ok
testSetProp (__main__.PyLibfdtBasicTests)
Test that we can update and create properties ... ok
testSetPropStr (__main__.PyLibfdtBasicTests)
Test that we can set a property to a particular string ... ok
testSetPropU32 (__main__.PyLibfdtBasicTests)
Test that we can update and create integer properties ... ok
testSetPropU64 (__main__.PyLibfdtBasicTests)
Test that we can update and create integer properties ... FAIL
testStrError (__main__.PyLibfdtBasicTests)
Check that we can get an error string ... ok
testSubnodeOffset (__main__.PyLibfdtBasicTests)
check that we can locate a subnode by name ... ok
testAccess (__main__.PyLibfdtRoTests)
Basic sanity check for the FdtRo class ... ok
testCreate (__main__.PyLibfdtSwTests) ... ERROR

======================================================================
ERROR: testCreate (__main__.PyLibfdtSwTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "pylibfdt_tests.py", line 517, in testCreate
    sw.property_u32('prop-int', TEST_VALUE_1)
  File "../pylibfdt/libfdt.py", line 913, in property_u32
    while self.check_space(fdt_property_u32(self._fdt, name, val)):
OverflowError: in method 'fdt_property_u32', argument 3 of type 'uint32_t'

======================================================================
FAIL: testReserveMap (__main__.PyLibfdtBasicTests)
Test that we can access the memory reserve map
----------------------------------------------------------------------
Traceback (most recent call last):
  File "pylibfdt_tests.py", line 406, in testReserveMap
    self.fdt.get_mem_rsv(0))
AssertionError: [16045690981097406464L, 1048576] != 1048576L

======================================================================
FAIL: testSetPropU64 (__main__.PyLibfdtBasicTests)
Test that we can update and create integer properties
----------------------------------------------------------------------
Traceback (most recent call last):
  File "pylibfdt_tests.py", line 458, in testSetPropU64
    self.fdt.getprop(node, prop))
AssertionError: '\xde\xad\xbe\xef\x01\xab\xcd\xef' != bytearray(b'\x00\x00\x00\x00\xff\xff\xff\xff')

----------------------------------------------------------------------
Ran 34 tests in 0.030s

FAILED (failures=2, errors=1)
********** TEST SUMMARY
*     Total testcases:	1796
*                PASS:	1793
*                FAIL:	1
*   Bad configuration:	0
* Strange test result:	0
**********
make: *** [tests/Makefile.tests:80: check] Error 1
````

dgibson, 2018-08-31:
Ugh, yeah.  It's broken on 32-bit x86 as well.  I've fixed one of the bugs, but the rest involves understanding swig's typemapping better than I do now.

sharkcz, 2018-08-31:
So the overflow error is away, thanks.

And I think I see what's wrong with the get_mem_rsv() - the memory map isn't even stored, likely due an overflow in the value, when the test is initialized, so later the test reads only a scalar value instead of a tuple.

closed 2018-09-02 by dgibson