dgibson/dtc #17: fs.test_tree1.test.dts aborted on ppc64 and s390x

sharkcz, 2018-09-26:
I get a test abort after commit 8c59a97ce096ab1e25fa828cbff371b3953e990c in our CI. It's possibly a big endian problem.
````
[sharkcz@tyan-openpower-01 dtc]$ coredumpctl gdb 10461
           PID: 10461 (dtc)
           UID: 1000 (sharkcz)
           GID: 1000 (sharkcz)
        Signal: 6 (ABRT)
     Timestamp: Wed 2018-09-26 10:40:30 CEST (21s ago)
  Command Line: ../dtc -I fs -O dts -o fs.test_tree1.test.dts fs/test_tree1
    Executable: /home/sharkcz/projects/dtc/dtc
 Control Group: /user.slice/user-1000.slice/session-39.scope
          Unit: session-39.scope
         Slice: user-1000.slice
       Session: 39
     Owner UID: 1000 (sharkcz)
       Boot ID: b0cee110e5ae499ba7f1d88e1f962ca6
    Machine ID: 46314d274cbd40d682e18963081ae0b7
      Hostname: tyan-openpower-01
       Storage: /var/lib/systemd/coredump/core.dtc.1000.b0cee110e5ae499ba7f1d88e1f962ca6.10461.1537951230000000.lz4
       Message: Process 10461 (dtc) of user 1000 dumped core.
                
                Stack trace of thread 10461:
                #0  0x00007fff8bf37a04 raise (libc.so.6)
                #1  0x00007fff8bf183a8 abort (libc.so.6)
                #2  0x00007fff8bf2b6b0 __assert_fail_base (libc.so.6)
                #3  0x00007fff8bf2b74c __assert_fail (libc.so.6)
                #4  0x000000001000f08c n/a (/home/sharkcz/projects/dtc/dtc)
                #5  0x000000001000f08c n/a (/home/sharkcz/projects/dtc/dtc)
                #6  0x000000001000f788 n/a (/home/sharkcz/projects/dtc/dtc)
                #7  0x0000000010002e14 n/a (/home/sharkcz/projects/dtc/dtc)
                #8  0x00007fff8bf18a08 generic_start_main.isra.0 (libc.so.6)
                #9  0x00007ffff12cf5e7 n/a (n/a)
                #10 0x00007ffff12cf5e7 n/a (n/a)
                #11 0x6f6e650053454c49 n/a (n/a)

GNU gdb (GDB) Fedora 8.1.1-3.fc28
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "ppc64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/sharkcz/projects/dtc/dtc...done.
[New LWP 10461]
Core was generated by `../dtc -I fs -O dts -o fs.test_tree1.test.dts fs/test_tree1 '.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007fff8bf37a04 in .raise () from /lib64/power8/libc.so.6
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.27-32.fc28.ppc64 libyaml-0.1.7-5.fc28.ppc64
(gdb) where
#0  0x00007fff8bf37a04 in .raise () from /lib64/power8/libc.so.6
#1  0x00007fff8bf183a8 in .abort () from /lib64/power8/libc.so.6
#2  0x00007fff8bf2b6b0 in .__assert_fail_base () from /lib64/power8/libc.so.6
#3  0x00007fff8bf2b74c in .__assert_fail () from /lib64/power8/libc.so.6
#4  0x000000001000f08c in write_propval_string (len=0, s=<optimized out>, f=<optimized out>) at treesource.c:67
#5  write_propval (prop=0x10039f08d10, f=0x10039f08330) at treesource.c:259
#6  write_tree_source_node (f=f@entry=0x10039f08330, tree=0x10039f082a0, level=level@entry=0) at treesource.c:294
#7  0x000000001000f788 in dt_to_source (f=0x10039f08330, dti=0x10039f08630) at treesource.c:321
#8  0x0000000010002e14 in main (argc=<optimized out>, argv=0x7ffff12cc078) at dtc.c:357
(gdb) q
````

sharkcz, 2018-09-26:
backtrace from -O0 build
````
(gdb) where
#0  0x00007ffff7d47a04 in .raise () from /lib64/power8/libc.so.6
#1  0x00007ffff7d283a8 in .abort () from /lib64/power8/libc.so.6
#2  0x00007ffff7d3b6b0 in .__assert_fail_base () from /lib64/power8/libc.so.6
#3  0x00007ffff7d3b74c in .__assert_fail () from /lib64/power8/libc.so.6
#4  0x0000000010012ab4 in write_propval_string (f=0x10058330, s=0x10058cf0 "hello world", len=0) at treesource.c:67
#5  0x0000000010013658 in write_propval (f=0x10058330, prop=0x10058d10) at treesource.c:259
#6  0x0000000010013918 in write_tree_source_node (f=0x10058330, tree=0x100582a0, level=0) at treesource.c:294
#7  0x0000000010013af0 in dt_to_source (f=0x10058330, dti=0x10058630) at treesource.c:321
#8  0x0000000010003210 in main (argc=8, argv=0x7fffffffe708) at dtc.c:357
````

dgibson, 2018-10-08:
I think it is fixed by 522d81d5

sharkcz, 2018-10-08:
yes, it is

closed 2018-10-08 by sharkcz