dgibson/dtc #19: libfdt: Import version v1.4.7 has bad impacts on performances

nlebayon, 2018-10-29:
On a ARM 32-bit platform, we are using Device tree and we observed a performance issue after the last rebase on libfdt 1.4.7.

For information, here are total times we spend in the FDT functions with different releases:
libfdt 1.4.2: 114ms
libfdt 1.4.6.9: 119ms (+4%)
libfdt 1.4.7: 653ms (+572%)

After a deeper analysis, we deduced that it was due to fdt32_ld / fdt64_ld function implementations, and more precisely to the memcpy operations inside, added for alignment reasons.

Note that on U-Boot community, as on Kernel one, they did not upgrade their FDT library on this 1.4.7 release, they are presently using libfdt 1.4.6.9. We can suppose that they didn't switch to the latest release due to this performance degradation, even if we were surprised not to find found any discussion on this subject here.

On our side, we had to modify these functions in order to remove the memcpy call thanks to a preprocessor conditional:
static inline uint32_t fdt32_ld(const fdt32_t *p)
{
fdt32_t __unused v;

#if FDT_NO_ALIGNMENT_CHECK
assert((*p & 0x3) == 0);
return fdt32_to_cpu(*p);
#else
memcpy(&v, p, sizeof(v));
return fdt32_to_cpu(v);
#endif
}

And the equivalent in fdt64_ld with a 0x7 mask.
With this modification, we recover more or less libfdt 1.4.2 performances (only +6% gap).

Did anyone else observed this kind of issue? We tried to give as much information as possible.

dgibson, 2018-11-19:
Ok, looks like those memcpy()s had a rather larger impact on performance than I'd anticipated.

However the approach above really isn't usable.  That handling for non-aligned values is to deal with conditions that really arise - and do so exactly on ARM machines which don't like unaligned accesses.

Unfortunately I don't really have access to ARM systems to observe this behaviour on.

So, I've made three temporary branches with three approaches which might help here (well, two approaches + the combination of both).  You cansee them at:
    https://github.com/dgibson/dtc/tree/unalign1
    https://github.com/dgibson/dtc/tree/unalign2
    https://github.com/dgibson/dtc/tree/unalign3

Could you see how those perform on you cases?

nlebayon, 2018-11-20:
Thanks David for your feedback.
I'm gonna test and measure performances with your three branches, and give you the results as soon as possible here.

nlebayon, 2018-11-21:
Hi David,
Here are the results with all the releases on our latest software. Note that timings are not exactly the same as above, but this is due to latest developments on our side. But anyway, you'll see the same trends.

libfdt 1.4.2: 111ms
libfdt 1.4.6.9: 114ms (+2.7%)
libfdt 1.4.7: 648ms (+483%)
unalign1: 214ms (+92%)
unalign2: 249ms (+124%)
unalign3: 242ms (+118%)

If you need more test or information, please do not hesitate.

dgibson, 2018-11-22:
So, unalign1 is the winner.  I'm pretty surprised by that, I thought it would be unalign2, and I'm especially surprised that unaligned 3 is slower than unalign1.  Oh well, I've made a proper commit message for unalign1 and merged it to master.

That may be the best we can accomplish reasonably.  It's still significantly slower than earlier versions, but that's due to fixes for real problems which I don't want to give up.  If you can see ways to speed it up that don't sacrifice too much else, I'm all ears.

closed 2018-11-22 by dgibson

nlebayon, 2018-11-22:
Following your feedback, I re-checked on my side, but these are the correct results on our ARM 32-bit platform. Indeed the difference is important, but we understand the interest of being concerned about alignment.

We will continue to think about an alternative solution, and we keep you informed. We think also that it could be interesting to have feedback from other platforms.

trini, 2020-01-16:
I hate to chime in on an old bug but this is all due to https://www.spinics.net/lists/devicetree-compiler/msg02230.html yes?  If so, the root cause was that the platform in question was doing something wrong by forcing everything to be unaligned to start with.  There's no need for these changes at all and we can go back to the old code.