dgibson/dtc #22: tests broken on ppc64/ppc64le

sharkcz, 2018-11-13:
commit baa1d2cf7894 breaks tests on ppc64/ppc64le, some segfault, some report "Bad configuration"
````
********** TEST SUMMARY
*     Total testcases:	1828
*                PASS:	1720
*                FAIL:	20
*   Bad configuration:	88
* Strange test result:	0
**********
````
full test log is in [dtc.log](https://github.com/dgibson/dtc/files/2577158/dtc.log)

dgibson, 2018-11-14:
Ah, damn.  I've reverted that now.  AFAICT it's bad on all platforms, but we get away with it by some accident on x86.  valgrind does detect the problem, even on x86, but there's something wrong in the scripts that meant it didn't fail the check in Travis, even though it displayed failures.

The config errors just looked to be due to missing input files which should have been generated by the dtc invocations that SEGVed.

closed 2018-11-14 by dgibson

sharkcz, 2018-11-14:
Thanks, all builds/tests are passing now again.