dgibson/dtc #26: fdt_overlay_apply fails on valid (?) ovelay blob

christophebraillon, 2019-02-28:
I encountered a problem using u-boot 2019.01 (which uses dtc DTC 1.4.6-gaadd0b65) on Raspberry Pi 3 while trying to apply some overlays to my base device tree. While some of them apply without any problem, others fail with error `FDT_ERR_BADOVERLAY.`

I tracked down the error to the function `overlay_symbol_update` line 797. The problem seems to be that this function expects a symbol do point to a subnode of an `__overlay__` node while my dts files name directly an `__overlay__` node and not one of its children.

You can have a look at [spi1-1cs-overlay.dts](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/spi1-1cs-overlay.dts) for example. The problem is generated by the symbol `frag1`.

The problem is that `dtc` authorizes this kind of naming an generates what I would consider a valid dtbo file regarding the device tree blob specifications, but then this dtbo generates an error when parsed.

What I did, for now, is modify my dts by changing the phandle in the `__overrides__` section this way:
```
	__overrides__ {
		cs0_pin  = <&spi1_cs_pins>,"brcm,pins:0",
			       <&spi1>,"cs-gpios:4";
		cs0_spidev = <&spidev1_0>,"status";
	};
```
That way I can remove the symbol `frag1`.

Wouldn't it be better to either modify dtc to raise an error when this kind of symbol is encountered or modify `overlay_symbol_update` to parse correctly those symbols?

dgibson, 2019-03-06:
Sorry I've taken so long to look at this.

So that I'm clear on your use case here, I think you're working with stacked overlays, yes?  You have a base tree, a first overlay which adds some symbols, and they you're going to want to apply a second overlay which uses symbols from the first overlay?

I don't think we'd considered the possibility of adding a label directly to the target of a fragment, which is effectively what this would do.  Looking at the code, I think making this would require a little fiddling, but shouldn't be too bad.

Would it be possible for you to write a (failing) testcase for libfdt that would test for the behaviour you want here?  That will make my life easier changing the libfdt code to handle it correctly.

christophebraillon, 2019-03-26:
Hi David,

Thanks for your answer, I beat you at late answer ;)

Actually I'm not trying to stack overlays. It fails when I try to apply a single overlay. I tried to make a test case so that you can figure out what to do. It is not really easy for me as kind of a newbie to the device tree world. I had a look at the test cases and I guess I reproduced my problem (I'm not entirely sure it fails at the same line in the code).

I just modified the file `stacked_overlay_baz.dts` that way:

```
/dts-v1/;
/plugin/;
/ {
	fragment@1 {
		target = <&bar>;
		toto: __overlay__ {
			overlay-2-property;
			baz: baznode {
				baz-property = "baz";
			};
		};
	};
};
```

I just name the `__overlay__` node in the tree. This results in the generation of new symbol in the dtbo file `toto: b'/fragment@1/__overlay__\x00'` which makes the `overlay_symbol_update` function fail at line 797 as it expects the symbol to have additionnal characters after `__overlay__`

I hope this will help.

dgibson, 2019-07-04:
I've finally had a chance to look at this.  As you say, we were outright rejecting these symbols, but that isn't really correct.  I now have a fix which should allow doing what you want in my master branch.

It also occurs to me that while you obviously couldn't merge a label like:
`foo: fragment@1 {
...
}`
Into the target tree, we probably shouldn't reject the overlay outright, but simply ignore that symbol.  I'm going to follow up with a fix for that.

closed 2019-07-04 by dgibson