dgibson/dtc #28: 0 is valid for size-cells in cpu node

yonghuhaige, 2019-07-25:
https://github.com/dgibson/dtc/blob/812b1956a07603fb56139f812395eb369727dd82/libfdt/fdt_addresses.c#L28

This bug is introduced in this commit, from v1.4.7, which will cause size-cells get failed.

```
SHA-1: c12b2b0c20eb8fbe1a22065f1d5f0f155ab56b0a

* libfdt: fdt_address_cells() and fdt_size_cells()

Add internal fdt_cells() to avoid copy and paste.  Test error cases and
default values.  Fix typo in fdt_size_cells() documentation comment.

```

Example dts file.
```
	#address-cells = <2>;
	#size-cells = <1>;
	cpus {
		#address-cells = <1>;
		#size-cells = <0>;
		cpu0: cpu@0 {
			compatible = "arm,cortex-a53", "arm,armv8";
			device_type = "cpu";
			logic_id = <0>;
			reg = <0x0>;
		};
		cpu1: cpu@1 {
			compatible = "arm,cortex-a53", "arm,armv8";
			device_type = "cpu";
			logic_id = <1>;
			reg = <0x1>;
		};
	};
```

yonghuhaige, 2019-07-25:
I'm not sure whether this file should be revert to previous version, or just modify as
```
if ((val < 0) || (val > FDT_MAX_NCELLS)) 
```

dgibson, 2019-07-28:
Ah, nice catch.  I've made a patch which I think should be the correct fix (also adding some tests to prevent regressions).  It's now committed to the master branch and will be in the next release.

closed 2019-07-28 by dgibson