dgibson/dtc #31: Cannot create a new property with fdt_appendprop()

AlexeiFedorov, 2019-12-05:
Hello, could you help me with the issue described below?
According to libfdt.h:
"fdt_appendprop - append to or create a property...
fdt_appendprop() appends the value to the named property in the
given node, creating the property if it does not already exist."

I'm trying to create a new property with a name of 14 bytes length in the existing node, but getting error -FDT_ERR_NOSPACE, which is returned by fdt_splice_() after failing check:
	if ((end - oldlen + newlen) > ((char *)fdt + fdt_totalsize(fdt))) {
		return -FDT_ERR_NOSPACE;
	}
Debug output:
ERROR:   fdt_splice_string_() fdt = 0x4001010 newlen 14
ERROR:    p = 0x4001226
ERROR:   fdt_splice_() fdt = 0x4001010 oldlen = 0 newlen 14
ERROR:    p 0x4001226 end 0x4001226 fdt_totalsize 534 fdt_data_size 534
ERROR:   FDT_ERR_NOSPACE

Does "if" statement fail because fdt_totalsize = fdt_data_size, or I'm doing something wrong here?
Thanks.

dgibson, 2019-12-06:
You probably need to call fdt_open_into() with a sufficiently large buffer before attempting to add content to the tree.

In order to add or extend properties with fdt_appendprop() (or any other libfdt function) there needs to be spare space in the blob to allow for it.  That is, the actual content of the tree needs to be smaller than fdt_totalsize().  If you've just read in a dtb from another source, that's probably not the case.  fdt_open_into() expands the the tree into a larger buffer, giving it the extra space you need for functions like fdt_appendprop().  You can either just expand once into a large enough buffer, or you can expand "on demand" by calling fdt_open_into() and retrying whenever a function fails with FDT_ERR_NOSPACE.

Once you're done editing, you can remove any leftover spare space with fdt_pack().

AlexeiFedorov, 2019-12-06:
Thank you, David! It was very helpful and fixed the issue.
I also found comment for FDT_ERR_NOSPACE in libfdt.h which explains that fdt_open_into() should be  called.

closed 2019-12-06 by AlexeiFedorov