dgibson/dtc #32: cannot static cross compile

lyq1996, 2020-03-01:
Hi,
Im using arm-linux-gnueabi-gcc to compile dtc, inorder to use it on TWRP recovery, but when I run 

```
~/dtc# make NO_PYTHON=1 CC=arm-linux-gnueabi-gcc dtc CFLAGS=-static -s
```
it always say
```
dtc.o: In function `main':
dtc.c:(.text+0xd00): undefined reference to `dt_to_yaml'
collect2: error: ld returned 1 exit status
Makefile:338: recipe for target 'dtc' failed
make: *** [dtc] Error 1
```
Could you please tell me what to do , thank you!!!

lyq1996, 2020-03-01:
The problem seems to resolve.

Here is my step:
```
make NO_PYTHON=1 CC=arm-linux-gnueabi-gcc
```
and then
```
~/dtc# make NO_PYTHON=1 CC=arm-linux-gnueabi-gcc CFLAGS=-static -s
```
then rm dtc and run again
```
~/dtc# make NO_PYTHON=1 CC=arm-linux-gnueabi-gcc CFLAGS=-static -s
~/dtc# file dtc
dtc: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), statically linked, stripped
```
Just don't know why..  
Sorry for bothering.
Thank you.

closed 2020-03-02 by lyq1996

dgibson, 2020-03-02:
Hm, I'm not really sure why that would be happening.

Do you have libyaml installed?  This might happen if the NO_YAML variable was somehow being set differently for building the .o files and the final link, but I don't know why that would happen.