dgibson/dtc #33: dts: appending to string property value?

cxw42, 2020-04-29:
First, thank you for dtc --- I would hate to hex-edit dtb files :) .

Is it possible to append to an existing string property value?  If not, could that option be added?

Use case: Nvidia Jetson Xavier, which reads its kernel command line from the dtb, node `/chosen/bootargs`.  At $work we have a common .dtsi file that is #included by individual .dts files for different configurations.  I would like to be able to set the bootargs in the dtsi file, then append to it in the dts files.  A dummy example follows.

My apologies if this function exists and I have not found it in my searching.  Thank you for considering this request!

common.dtsi:

    /dts-v1/;
    / { chosen { bootargs = "root=/dev/mmcblk0p1"; }; };

particular.dts:

    #include "common.dtsi"
    / { chosen { bootargs += " console=ttyTCU0,115200n8"; }; };

TheSven73, 2020-05-29:
If you need this new function just to set the debug console in your particular.dts, consider using [`stdout-path`](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/Documentation/devicetree/bindings/chosen.txt?h=v5.6.15#n31).

cxw42, 2020-05-29:
@TheSven73 thanks for the tip!  I was trying to provide a minimal example, but in my actual use case we are using custom bootargs read by a bespoke driver.  I will keep `stdout-path` in mind for the future, though!

TheSven73, 2020-05-29:
Hmm... instead of letting your driver read bootargs, you could have it look for a specific node included by your particular.dts. Or better still, let particular.dts add to the driver's own node, if it exists.

But I have the feeling you must have considered this already, and found wanting...

dgibson, 2021-02-17:
On Wed, Apr 29, 2020 at 05:39:31PM +0000, Chris White wrote:
> First, thank you for dtc --- I would hate to hex-edit dtb files :) .
> 
> Is it possible to append to an existing string property value?

I'm afraid not.

>If not, could that option be added?

Theoretically, yes.  It would be easily to hack in, but much, much
harder to implement in a way that has solid semantics that don't
conflict with other things.

> Use case: Nvidia Jetson Xavier, which reads its kernel command line from the dtb, node `/chosen/bootargs`.  At $work we have a common .dtsi file that is #included by individual .dts files for [...]

Honestly, for this simple case, I think a much more feasible approach
would be to write a small helper program using libfdt that does the
append on an already-compiled .dtb.

-- 
David Gibson			| I'll have my music baroque, and my code
david AT gibson.dropbear.id.au	| minimalist, thank you.  NOT _the_ _other_
				| _way_ _around_!
http://www.ozlabs.org/~dgibson

cxw42, 2021-02-17:
@dgibson thanks for getting back to me!  Since I opened this issue, NVIDIA has changed the Xavier boot path to use U-boot+extlinux, so we can now append command-line args in extlinux.conf.  Since there's a viable workaround and appending would be nontrivial to add to dtc and the spec, I will withdraw this request.

closed 2021-02-17 by cxw42