dgibson/dtc #39: fdt_next_node() return value depends on VALID_INPUT

frankmehnert, 2020-08-06:
The `fdt_next_node()` function calls `fdt_check_node_offset_()` to verify if the resulting offset is valid. The function `fdt_check_node_offset_()` returns without any further check if the `VALID_INPUT` flag is set at `FDT_ASSUME_MASK`.

If the `VALID_INPUT` flag is not set then fdt_check_node_offset_() calls
```
fdt_next_tag(fdt, offset, &offset)
```
and passes the returned offset to the caller. This offset can be different from the original offset.

I'm trying to iterate over a tree using
```
fdt_for_each_subnode(node, fdt, parent)
```
This works only if I disable `VALID_INPUT`.

dgibson, 2020-08-15:
Can you confirm if the latest push fixes this?  If so we can close this issue.

frankmehnert, 2020-08-15:
Confirmed. Thanks!

closed 2020-08-15 by frankmehnert