dgibson/dtc #42: Doesn't support nested device trees?

tomtastic, 2020-09-29:
I have a FDT blob which contains three sub tree's: 
```
/dts-v1/;
fdts {
   fdt@1 {
      platform_name = "imx6sll-a.dtb";
      property1 = "blah";
      property2 = "blah";
      property3 = "blah";
   fdt@2 {
      platform_name = "imx6sll-b-wfo.dtb";
      property1 = "blah";
      property2 = "blah";
      property3 = "blah";
   fdt@3 {
      platform_name = "imx6sll-b-wan.dtb";
      property1 = "blah";
      property2 = "blah";
      property3 = "blah";```

If I try to convert it to string format, all the properties of each of the subtree's get converted to a giant array of bytes :
```
/dts-v1/;

/ {

        fdts {

                fdt@1 {
                        platform_name = "imx6sll-a.dtb";
                        data = [d0 0d fe ed 00 00 76 8b 00 00 00 38 00 00 6e 28 00 00 00 28 00 00 00 11 00 00 00 10 00 00 00 00 00 00 08 63 00 00 6d f0 00 00 00 00 etc etc
```

Could there be support for such cases added ?

dgibson, 2020-09-30:
I've really no idea what you're trying to do here.  "Nested device trees" isn't really a thing.

tomtastic, 2020-10-06:
So I have a device tree file :
```
$ binwalk fdt_3objects.dtb
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Flattened device tree, size: 92335 bytes, version: 17
128           0x80            Flattened device tree, size: 30347 bytes, version: 17
30544         0x7750          Flattened device tree, size: 30596 bytes, version: 17
61208         0xEF18          Flattened device tree, size: 31089 bytes, version: 17
```

It appears to contain at least three flattened device trees, with what appear to be various properties visible at strings in each:
```
$ grep -A9 'fdt@' <(strings fdt_3objects.dtb)
fdt@1
imx6sll-rex.dtb
Lab126 i.MX6SLL Board
!amzn,imx6sll-rex
fsl,imx6sll
chosen
aliases
,/soc/aips-bus@02000000/gpio@0209c000
2/soc/aips-bus@02000000/gpio@020a0000
8/soc/aips-bus@02000000/gpio@020a4000
--
fdt@2
imx6sll-moonshine-wfo.dtb
Lab126 i.MX6SLL Board
!amzn,imx6sll-rex
fsl,imx6sll
chosen
aliases
,/soc/aips-bus@02000000/gpio@0209c000
2/soc/aips-bus@02000000/gpio@020a0000
8/soc/aips-bus@02000000/gpio@020a4000
--
fdt@3
imx6sll-moonshine-wan.dtb
Lab126 i.MX6SLL Board
!amzn,imx6sll-rex
fsl,imx6sll
chosen
aliases
,/soc/aips-bus@02000000/gpio@0209c000
2/soc/aips-bus@02000000/gpio@020a0000
8/soc/aips-bus@02000000/gpio@020a4000
```

If I attempt to convert it into a DTS file, those text strings disappear into some kind of long 'data' array of hex bytes :
```
$ dtc -I dtb -O dts -o - fdt_3objects.dtb | head -10
<stdout>: Warning (unit_address_vs_reg): /fdts/fdt@1: node has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): /fdts/fdt@2: node has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): /fdts/fdt@3: node has a unit name, but no reg property
/dts-v1/;

/ {

	fdts {

		fdt@1 {
			platform_name = "imx6sll-rex.dtb";
			data = [d0 0d fe ed 00 00 76 8b 00 00 00 38 00 00 6e 28 00 00 00 28 00 00 00 ... etc etc etc
```

I hope that clarifies things, as I was expecting to see a more familiar output, something like this :
```
    model = "Lab126 i.MX6SLL Board";
    compatible = "amzn,imx6sll-rex", "fsl,imx6sll";
    chosen {
    };
    aliases {
        gpio0 = "/soc/aips-bus@02000000/gpio@0209c000";
        gpio1 = "/soc/aips-bus@02000000/gpio@020a0000";
        gpio2 = "/soc/aips-bus@02000000/gpio@020a4000";
        etc...
```

dgibson, 2020-10-07:
On Tue, Oct 06, 2020 at 07:18:53AM -0700, Tom Matthews wrote:
> So I have a device tree file :
> ```
> $ binwalk fdt_3objects.dtb
> DECIMAL       HEXADECIMAL     DESCRIPTION
> --------------------------------------------------------------------------------
> 0 [...]

Ah, ok.  I think that's a format the uboot people made to bundle a
bunch of DTs (and sometimes kernels) together.  Although it's using
the dtb encoding, the outer thing isn't really a "device tree"
itself.  The inner device trees are indeed just binary device tree
blobs stored in properties (properties are just bytestrings, so that's
easy).

Since this isn't really part of the device tree spec as such, I don't
think it really makes sense to have explicit support for this in dtc.

I suggest you use 'fdtget' to extract the relevant properties with the
contained device trees, then you can use dtc to decode those.

-- 
David Gibson			| I'll have my music baroque, and my code
david AT gibson.dropbear.id.au	| minimalist, thank you.  NOT _the_ _other_
				| _way_ _around_!
http://www.ozlabs.org/~dgibson

tomtastic, 2020-10-07:
Thank you for the explanation, I think we may as well close this issue then.

closed 2020-10-07 by tomtastic