dgibson/dtc #45: FATAL ERROR: String offset overruns string table

dchard, 2021-01-01:
Hi,

I am trying to convert a DTB file of a Xiaomi AX6 router to DTS, and I get this message:

```
root@nas:~/dtc# ./dtc -I dtb -O dts /root/xiaomi_ax6/AP-AC04.dtb
FATAL ERROR: String offset 16777216 overruns string table
```

As much as I can tell, the DTB file looks valid. I attached the affected DTB file:
[AP-AC04.ZIP](https://github.com/dgibson/dtc/files/5759341/AP-AC04.ZIP)

dgibson, 2021-01-02:
The dtb appears to be badly corrupted.  However you've extracted it, it's been damaged somehow.

dchard, 2021-01-02:
> 
> 
> The dtb appears to be badly corrupted. However you've extracted it, it's been damaged somehow.

Thanks for checking this! I used [dtbextract](https://github.com/PabloCastellano/extract-dtb) script to extract it from the firmware image, and this is the one used by the router. There was another dtb file extracted in the same process and that can be converted to dts with dtc.

Anyway, if you confirm that the DTB is corrupt, then this is not a DTC issue, so I will close this issue.

closed 2021-01-02 by dchard