dgibson/dtc #48: Line numbers?

ab1jx, 2021-04-12:
It would be really helpful if errors like "FATAL ERROR: Unable to parse input tree" could include a line number where the error is.  I'm trying to work with a dts file I didn't make myself that's over 1000 lines long and I have no idea where to look.

ab1jx, 2021-04-13:
It seems it does give a line number, I thought it was some version number.
Error: rk3399-pinebook-pro.dts:9.1-9  has an error in line 9, the first number after the colon.  But if you feed it some junk file the number changes to line 1.  And it has to go through cpp first which I didn't realize, some documentation out there implies you can run it without.

closed 2021-04-13 by ab1jx

dgibson, 2021-04-13:
Right, that way of formatting line (and column) information is borrowed from the way gcc displays errors, IIRC.

dtc certainly doesn't require its input to go through cpp, though your particular dts file might.

ab1jx, 2021-04-13:
If the dts file you want to use has includes you're stuck with cpp I
think.  It was simpler to borrow a dtb file from a different Linux for
the same hardware.