dgibson/dtc #49: fdtput does not seem to support to add a new node as a children

MagicMuscleMan, 2021-05-01:
Either it's not documented or it's not supported to add a new node as a children using fdtput. This limits the use of fdtput.

dgibson, 2021-05-04:
I'm not sure exactly what you're asking for here.  Do you mean you want `fdtput` to create a new node (and any necessary parent nodes) if asked to add a property to one that doesn't exist?

I think that case just wasn't considered when writing it.  Patches welcome?

MagicMuscleMan, 2021-05-04:
Yes, the problem is you can use fdtput to extend the device tree entry by entry when calling it multiple times. However, I was not able to create a new node (which is per default a child node to another node if I understood the device tree specification correctly). Therefore I couldn't use fdtput to solve my problem.

I can think of two interfaces:

- Add a new type to the -t parameter (n=node) to explicitly create a new node in a separate fdtput call
- Create the new node implicitly when a property is set for a node that does not exist (recursively, as you described)

However, realistically, I won't create a patch to provide this functionality, as it was only a side problem and I basically stopped using fdtput directly after starting to use it. So please feel free to close this bug, if you think this is the best way to handle it in this situation.

dgibson, 2021-05-13:
Right, those are the two obvious interfaces, or alternatively adding a new fdtsubnode command.

It's a reasonable thing to want, and it shouldn't be that complicated but I don't have the spare time to work on it, so I don't expect it to happen until/unless someone volunteers to do it.

dgibson, 2026-10-28:
Turns out this functionality already existed.  I just merged a patch from @sjg20 which improves the documentation around it.

closed 2026-10-28 by dgibson