dgibson/dtc #52: valid_header function may be have some problem.

juiceRv, 2021-08-10:
https://github.com/dgibson/dtc/blob/ecfb438c07fa468a129e81c7d84c7c293c7b0150/fdtdump.c#L166
This function should be modified like this?
```
static bool valid_header(char *p, size_t len)
{
	if (len < sizeof(struct fdt_header) ||
	    fdt_magic(p) != FDT_MAGIC ||
	    fdt_version(p) > MAX_VERSION ||
	    fdt_last_comp_version(p) > MAX_VERSION ||
	    fdt_totalsize(p) > len ||
	    fdt_off_dt_struct(p) > len ||
	    fdt_off_dt_strings(p) > len)
		return 0;
	else
		return 1;
}
```

juiceRv, 2021-08-10:
i look some code like this https://github.com/torvalds/linux/blob/5bfc75d92efd494db37f5c4c173d3639d4772966/scripts/dtc/flattree.c#L876

dgibson, 2023-02-02:
@juiceRv sorry, I don't understand what change you're suggesting.  The code you quote appears to be identical to what's in dtc now.

dgibson, 2023-02-27:
No response to query, so closing.

closed 2023-02-27 by dgibson