dgibson/dtc #58: pylibfdt fails to work with 3.10

rossburton, 2021-11-11:
```
ERROR: testCreate (__main__.PyLibfdtSwTests)
----------------------------------------------------------------------
SystemError: PY_SSIZE_T_CLEAN macro must be defined for '#' formats
```

`PY_SSIZE_T_CLEAN` is now mandatory in 3.10, so the module needs to define that symbol and change the length type from `int` to `ssize_t` in all `#` formats passed to eg `Py_BuildValue`.

I'm new to Swig so trying to figure out how to make this actually happen...

rossburton, 2021-11-11:
So Python expects the length, or `*arg4` in the `%typemap(out) (const void *)` typemap, to be `Py_ssize_t`.  The API uses `int` so it needs to be cast.

rossburton, 2021-11-11:
Made it work.  Define PY_SSIZE_T_CLEAN, cast the ints to Py_ssize_t.  Patch incoming once I've cleared it internally.

dgibson, 2021-11-12:
Thanks for your patch, that's merged now, so I'm closing.

closed 2021-11-12 by dgibson