dgibson/dtc #59: `meson test` doesn't execute Python test suite

rossburton, 2021-11-11:
If I use the Meson build, `meson test` doesn't actually run the Python tests.

dgibson, 2021-11-12:
Uh... as far as I can tell, it is running the Python tests, it's just not obvious, because the testsuite is not "meson native".

I get:
```
$ meson test
ninja: Entering directory `/home/dwg/src/dtc/build'
ninja: no work to do.
1/1 run-test        OK                2.39s

Ok:                 1   
Expected Fail:      0   
Fail:               0   
Unexpected Pass:    0   
Skipped:            0   
Timeout:            0   

Full log written to <redacted>/dtc/build/meson-logs/testlog.txt
```

But if I look at the testlog.txt, it ends with:

```
Test that we can update and create integer properties ... ok
testStrError (__main__.PyLibfdtBasicTests)
Check that we can get an error string ... ok
testSubnodeOffset (__main__.PyLibfdtBasicTests)
check that we can locate a subnode by name ... ok
testAccess (__main__.PyLibfdtRoTests)
Basic sanity check for the FdtRo class ... ok
testCreate (__main__.PyLibfdtSwTests) ... ok

----------------------------------------------------------------------
Ran 36 tests in 0.003s

OK
********** TEST SUMMARY
*     Total testcases:  2085
*                PASS:  2085
*                FAIL:  0
*   Bad configuration:  0
* Strange test result:  0
**********
```

Which shows the python tests running, and the right total number of tests for including all the python tests.

Using the meson native test infrastructure would be much nicer, but that's a larger project, and I don't really have the meson knowledge to do it at the moment.

rossburton, 2021-11-12:
Yes, you're right.

closed 2021-11-12 by rossburton