dgibson/dtc #60: Python test suite fails

rossburton, 2021-11-11:
With current master the Python test suite fails:

```
FAIL: testHeader (__main__.PyLibfdtBasicTests)
Test that we can access the header values
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./pylibfdt_tests.py", line 295, in testHeader
    self.assertEqual(self.fdt.size_dt_strings(), 105)
AssertionError: 97 != 105

```

dgibson, 2021-11-12:
Is this related to the PY_SSIZE_T thing?  This is a different error, but now that I've applied that fix, the testsuite is passing for me.

rossburton, 2021-11-12:
No, it's fixed by "pylibfdt_tests: Don't depend on built dtbs" patch from Rob.  Depending on how you run the tests changes the DTBs that are used.

closed 2023-02-02 by dgibson