dgibson/dtc #64: A possible case to hang the compiler

Escapingbug, 2021-12-11:
```
/dts-v1/;

/ {
    clocks = <&ref &ref>;

    ref: poc {
        phandle = <1>;
        #clock-cells = <0xffffffff>;
    };
};
```

During checks, in function [`check_property_phandle_args`](https://github.com/dgibson/dtc/blob/main/checks.c#L1384), the `0xffffffff` value could trigger an integer overflow which will keep us in the loop forever.

dgibson, 2021-12-29:
Thanks for the report.  Sorry it's taken me so long to look at it.  Certainly looks like a bug, I've made an automated test case and I'm investigating now.

dgibson, 2021-12-29:
Ok, I've applied a fix, it would be great if you can check it.

dgibson, 2023-02-02:
Closing, since there's been no response.

closed 2023-02-02 by dgibson