dgibson/dtc #75: New release planned?

a3f, 2023-01-27:
v1.6.1 was released on June 8th, 2021. Are there plans for a new release in near future? Are there blockers I could help with?

dgibson, 2023-01-28:
On Fri, Jan 27, 2023 at 10:07:04AM -0800, Ahmad Fatoum wrote:
> v1.6.1 was released on June 8th, 2021. Are there plans for a new
> release in near future?

I don't really plan releases, but several people have requested one.

> Are there blockers I could help with?

Kind of, yes.  On my system, at least, the Python library doesn't
build/install, with a cryptic error in setuptools.  That's the main
blocker to making a release right now.

-- 
David Gibson			| I'll have my music baroque, and my code
david AT gibson.dropbear.id.au	| minimalist, thank you.  NOT _the_ _other_
				| _way_ _around_!
http://www.ozlabs.org/~dgibson

dgibson, 2023-02-02:
See #78 for more info on the pylibfdt build failure.

dgibson, 2023-02-13:
@a3f #78 turns out not to have been a real bug (well, mostly) but something messed up in my local git tree.  So, I have that working again.  Next, I have some problems with updates due to problems with my kernel.org public key.  I'm working on that and hope to have it sorted out soon.

a3f, 2023-02-21:
Thanks for the release! [Already imported into barebox-next](https://lore.barebox.org/barebox/20230221102506.GE10447@pengutronix.de/T/#mea7c53ba5ee110774ee7e5d06a63c945c6d11f18). :-)

dgibson, 2023-02-28:
Unfortunately there are some problems in the release - it took me a while to get an actual signed tarball up, and then I discovered I forgot to bump the version for meson builds.  So, I probably need to make a 1.7.1 pretty soon.  Still I hope this works for you in the meantime.

eli-schwartz, 2023-03-01:
> and then I discovered I forgot to bump the version for meson builds

Maybe it's worth single-sourcing the version number? You can use something like `VERSION.txt` and read it into meson with:
```diff
-  version: '1.7.0',
+  version: files('VERSION.txt'),
```

I'd say you could then read it in the Makefile and split it into 3 (`subst`, `word n` for example) but it doesn't seem you need it to be 3 different Makefile variables anyway?

eli-schwartz, 2023-03-01:
Note that using files() like this requires a minimum meson version of 0.57.0 (no `meson_version:` is specified in meson.build though).

dgibson, 2023-03-01:
That's obviously the best approach, it's just a question of finding time to do it.