dgibson/dtc #77: last commit breaks building the Python bindings

sharkcz, 2023-01-30:
Seems the last commit a41509bea3e73b8dc63bf666a82275bc7d040266 breaks building the Python bindings.
````
...
make[1]: Entering directory '/var/lib/jenkins/workspace/dtc/label/x86_64'
set -e; echo '	CHK version_gen.h'; mkdir -p ./; 	(echo "#define DTC_VERSION \"DTC 1.6.1-ga41509be\""; ) < Makefile > version_gen.h.tmp; if [ -r version_gen.h ] && cmp -s version_gen.h version_gen.h.tmp; then rm -f version_gen.h.tmp; else echo '	UPD version_gen.h'; mv -f version_gen.h.tmp version_gen.h; fi;
	CHK version_gen.h
python3 ./setup.py  build_ext --build-lib=pylibfdt
running build_ext
building '_libfdt' extension
swigging /var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt.i to /var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c
swig -python -I/var/lib/jenkins/workspace/dtc/label/x86_64/./libfdt -o /var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c /var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt.i
creating build
creating build/temp.linux-x86_64-3.10
creating build/temp.linux-x86_64-3.10/var
creating build/temp.linux-x86_64-3.10/var/lib
creating build/temp.linux-x86_64-3.10/var/lib/jenkins
creating build/temp.linux-x86_64-3.10/var/lib/jenkins/workspace
creating build/temp.linux-x86_64-3.10/var/lib/jenkins/workspace/dtc
creating build/temp.linux-x86_64-3.10/var/lib/jenkins/workspace/dtc/label
creating build/temp.linux-x86_64-3.10/var/lib/jenkins/workspace/dtc/label/x86_64
creating build/temp.linux-x86_64-3.10/var/lib/jenkins/workspace/dtc/label/x86_64/pylibfdt
gcc -Wno-unused-result -Wsign-compare -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -fPIC -DPY_SSIZE_T_CLEAN -I/var/lib/jenkins/workspace/dtc/label/x86_64/./libfdt -I/usr/include/python3.10 -c /var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c -o build/temp.linux-x86_64-3.10/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.o
/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c: In function ‘_wrap_fdt_node_header_name_set’:
/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c:4038:7: warning: the comparison will always evaluate as ‘true’ for the address of ‘name’ will never be NULL [-Waddress]
 4038 |   if (arg1->name) free((char*)arg1->name);
      |       ^~~~
In file included from /var/lib/jenkins/workspace/dtc/label/x86_64/./libfdt/libfdt.h:10,
                 from /var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c:2722:
/var/lib/jenkins/workspace/dtc/label/x86_64/./libfdt/fdt.h:38:14: note: ‘name’ declared here
   38 |         char name[];
      |              ^~~~
/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c:4041:18: error: cast specifies array type
 4041 |     arg1->name = (char [])(char *)memcpy(malloc((size)*sizeof(char)), (const char *)(arg2), sizeof(char)*(size));
      |                  ^
/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c:4041:16: error: invalid use of flexible array member
 4041 |     arg1->name = (char [])(char *)memcpy(malloc((size)*sizeof(char)), (const char *)(arg2), sizeof(char)*(size));
      |                ^
/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c:4043:16: error: invalid use of flexible array member
 4043 |     arg1->name = 0;
      |                ^
/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c: In function ‘_wrap_fdt_property_data_set’:
/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c:4301:7: warning: the comparison will always evaluate as ‘true’ for the address of ‘data’ will never be NULL [-Waddress]
 4301 |   if (arg1->data) free((char*)arg1->data);
      |       ^~~~
/var/lib/jenkins/workspace/dtc/label/x86_64/./libfdt/fdt.h:45:14: note: ‘data’ declared here
   45 |         char data[];
      |              ^~~~
/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c:4304:18: error: cast specifies array type
 4304 |     arg1->data = (char [])(char *)memcpy(malloc((size)*sizeof(char)), (const char *)(arg2), sizeof(char)*(size));
      |                  ^
/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c:4304:16: error: invalid use of flexible array member
 4304 |     arg1->data = (char [])(char *)memcpy(malloc((size)*sizeof(char)), (const char *)(arg2), sizeof(char)*(size));
      |                ^
/var/lib/jenkins/workspace/dtc/label/x86_64/./pylibfdt/libfdt_wrap.c:4306:16: error: invalid use of flexible array member
 4306 |     arg1->data = 0;
      |                ^
error: command '/usr/bin/gcc' failed with exit code 1
make[1]: *** [pylibfdt/Makefile.pylibfdt:20: pylibfdt/_libfdt.so] Error 1
make[1]: Leaving directory '/var/lib/jenkins/workspace/dtc/label/x86_64'
make: *** [Makefile:172: maybe_pylibfdt] Error 2
Build step 'Execute shell' marked build as failure
Finished: FAILURE
````
This is on Fedora 36 on x86_64, but fails on all arches.

dgibson, 2023-02-02:
Someone else also reported this, and a fix has been applied (commit abbd523bae6).

> This is on Fedora 36 on x86_64, but fails on all arches.

Huh.  That's interesting.  The reason I didn't notice this breakage before applying is that build of pylibfdt has been broken for me for some time.  It's not broken for everyone, so it's something specific to my build environment - which is Fedora 37, and I'm pretty sure it was also broken when I was on Fedora 36.  So now I'm even more at a loss as to why it's broken for me.

closed 2023-02-02 by dgibson

sharkcz, 2023-02-06:
for the record, all looks good again after commit a41509bea3e73b8dc63bf666a82275bc7d040266