dgibson/dtc #78: pylibfdt bindings cryptically fail to build with some build environments

dgibson, 2023-02-02:
I've been unable to build the pylibfdt bindings for some time.   When I try I get this error:

```
$ make
	CHK version_gen.h
make[1]: Entering directory '/home/dwg/src/dtc'
	CHK version_gen.h
	 PYMOD pylibfdt/_libfdt.so
/usr/lib/python3.11/site-packages/setuptools_scm/version.py:84: UserWarning: tag 'path-ref-fixup-staging' no version found
  warnings.warn(f"tag {tag!r} no version found")
Traceback (most recent call last):
  File "/home/dwg/src/dtc/./setup.py", line 50, in <module>
    setup(
  File "/usr/lib/python3.11/site-packages/setuptools/__init__.py", line 87, in setup
    return distutils.core.setup(**attrs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 139, in setup
    _setup_distribution = dist = klass(attrs)
                                 ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools/dist.py", line 477, in __init__
    _Distribution.__init__(
  File "/usr/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 275, in __init__
    self.finalize_options()
  File "/usr/lib/python3.11/site-packages/setuptools/dist.py", line 901, in finalize_options
    ep(self)
  File "/usr/lib/python3.11/site-packages/setuptools/dist.py", line 922, in _finalize_setup_keywords
    ep.load()(self, ep.name, value)
  File "/usr/lib/python3.11/site-packages/setuptools_scm/integration.py", line 90, in version_keyword
    _assign_version(dist, config)
  File "/usr/lib/python3.11/site-packages/setuptools_scm/integration.py", line 59, in _assign_version
    maybe_version = _get_version(config)
                    ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools_scm/__init__.py", line 154, in _get_version
    parsed_version = _do_parse(config)
                     ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools_scm/__init__.py", line 101, in _do_parse
    version = _version_from_entrypoints(config) or _version_from_entrypoints(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools_scm/_entrypoints.py", line 66, in _version_from_entrypoints
    version: ScmVersion | None = _call_entrypoint_fn(root, config, ep.load())
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools_scm/_entrypoints.py", line 40, in _call_entrypoint_fn
    return fn(root, config=config)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools_scm/git.py", line 181, in parse
    return _git_parse_inner(
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools_scm/git.py", line 224, in _git_parse_inner
    return meta(
           ^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools_scm/version.py", line 226, in meta
    parsed_version = _parse_tag(tag, preformatted, config)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools_scm/version.py", line 203, in _parse_tag
    assert version is not None
AssertionError
make[1]: *** [pylibfdt/Makefile.pylibfdt:20: pylibfdt/_libfdt.so] Error 1
make[1]: Leaving directory '/home/dwg/src/dtc'
make: *** [Makefile:172: maybe_pylibfdt] Error 2
```

Build environment is Fedora 37:

- python3-3.11.1-3.fc37.x86_64
- python-setuptools-wheel-62.6.0-2.fc37.noarch
- python3-setuptools-62.6.0-2.fc37.noarch
- python3-setuptools_scm-7.0.5-1.fc37.noarch

a3f, 2023-02-11:
I failed to reproduce this in a Fedora 37 podman container:
```
podman run -it fedora:37 /bin/bash
dnf install python3-3.11.1-3.fc37.x86_64
dnf install python-setuptools-wheel-62.6.0-2.fc37.noarch
dnf install python3-setuptools-62.6.0-2.fc37.noarch
dnf install python3-setuptools_scm-7.0.5-1.fc37.noarch 
dnf install diffutils git make gcc swig flex bison python3-devel which
git clone https://github.com/dgibson/dtc
cd dtc
make
```
And it runs to completion:
```
make[1]: Entering directory '/root/dtc'
	CHK version_gen.h
	PYMOD pylibfdt/_libfdt.so
make[1]: Leaving directory '/root/dtc'
```

dgibson, 2023-02-13:
Looks like this is triggered by something odd in my local git tree - the problem goes away with a new clone.

Trying to work out what exactly is triggering the problem, still.

dgibson, 2023-02-27:
Honestly, I'm unlikely to ever get the time to track this down.  I'll just use by new tree and hope for the best.

closed 2023-02-27 by dgibson