dgibson/dtc #83: README.md has no instructions for building with meson

dgibson, 2023-02-27:
We currently have both Makefiles and meson scripts either of which can build the project.  However, I frequently forget how to properly invoke meson, so it tends to be poorly tested.  `README.md` should have short instructions for how to build with meson.

z3ntu, 2023-05-02:
I guess not really more than the few commands here are necessary? https://mesonbuild.com/Quick-guide.html#compiling-a-meson-project

And maybe link people to the docs there for things like advanced usage with DESTDIR etc

dgibson, 2023-05-03:
Yes.  The shorter the better, really.

sjg20, 2026-10-28:
I sent a few patches for this:

https://lore.kernel.org/devicetree-compiler/20261028112007.3980089-1-sjg@chromium.org/T/#t

closed 2026-10-30 by sjg20