dgibson/dtc #90: 'struct fdt_header' has no member named [name]

Chan-I, 2023-04-10:
I am just trying to build the OpenBMC and met some errors: 
```
| libfdt/libfdt.h:251:28: error: array subscript ‘struct fdt_header[0]’ is partly outside array bounds of ‘unsigned char[4]’ [-Werror=array-bounds]
|   251 |                 fdth->name = cpu_to_fdt32(val); \
|       |                 ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
| libfdt/libfdt.h:253:1: note: in expansion of macro ‘fdt_set_hdr_’
|   253 | fdt_set_hdr_(magic);
|       | ^~~~~~~~~~~~
| fdtdump.c:216:31: note: while referencing ‘smagic’
|   216 |                 unsigned char smagic[FDT_MAGIC_SIZE];
|       |                               ^~~~~~
| cc1: all warnings being treated as errors
```

After reading the code of this project, I have some questions.

The definition of ```struct fdt_header``` in ```dtc/fdt.h``` is as follows,
It does not have a member named [name].
```c
struct fdt_header {
	fdt32_t magic;			 /* magic word FDT_MAGIC */
	fdt32_t totalsize;		 /* total size of DT block */
	fdt32_t off_dt_struct;		 /* offset to structure */
	fdt32_t off_dt_strings;		 /* offset to strings */
	fdt32_t off_mem_rsvmap;		 /* offset to memory reserve map */
	fdt32_t version;		 /* format version */
	fdt32_t last_comp_version;	 /* last compatible version */

	/* version 2 fields below */
	fdt32_t boot_cpuid_phys;	 /* Which physical CPU id we're
					    booting on */
	/* version 3 fields below */
	fdt32_t size_dt_strings;	 /* size of the strings block */

	/* version 17 fields below */
	fdt32_t size_dt_struct;		 /* size of the structure block */
};
```

But in  ```dtc/libfdt.h : 264```
```c
260:#define fdt_set_hdr_(name) \
261:	static inline void fdt_set_##name(void *fdt, uint32_t val) \
262:	{ \
263:		struct fdt_header *fdth = (struct fdt_header *)fdt; \
264:		fdth->name = cpu_to_fdt32(val); \
265:	}
```

Is that a BUG?

closed 2023-04-10 by Chan-I

WiFiSmashr, 2023-07-12:
Hello, I am getting a similar error message as you at the top of this post. I was wondering if you had any information on how you solved your issue?

dgibson, 2023-07-19:
The `name` here is the macro parameter.  The macro is being used to generate the actual accessor inline functions.  This error appears to be the result of the macro not being expanded properly.  I'm guessing that's something due to how the code is appearing in context in OpenBMC.  Alternatively it could be that the OpenBMC build is using a different compiler to those I've tested.