dgibson/dtc #91: 1.7.0: install target fails on

kloczek, 2023-04-13:
`setuptools`  67.6.1.

meson and meson build output:
<details>

```console
+ /usr/bin/meson setup --buildtype=plain --prefix=/usr --libdir=/usr/lib64 --libexecdir=/usr/libexec --bindir=/usr/bin --sbindir=/usr/sbin --includedir=/usr/include --datadir=/usr/share --mandir=/usr/share/man --infodir=/usr/share/info --localedir=/usr/share/locale --sysconfdir=/etc --localstatedir=/var --sharedstatedir=/var/lib --wrap-mode=nodownload --auto-features=enabled . x86_64-redhat-linux-gnu -D python=enabled -D tools=true -D valgrind=enabled -D yaml=enabled
The Meson build system
Version: 1.1.0
Source dir: /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0
Build dir: /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/x86_64-redhat-linux-gnu
Build type: native build
Project name: dtc
Project version: 1.6.0
C compiler for the host machine: /usr/bin/gcc (gcc 13.0.1 "gcc (GCC) 13.0.1 20230404 (Red Hat 13.0.1-0)")
C linker for the host machine: /usr/bin/gcc ld.bfd 2.40-2
Host machine cpu family: x86_64
Host machine cpu: x86_64
Compiler for C supports arguments -Wpointer-arith: YES
Compiler for C supports arguments -Wcast-qual: YES
Compiler for C supports arguments -Wnested-externs: YES
Compiler for C supports arguments -Wstrict-prototypes: YES
Compiler for C supports arguments -Wmissing-prototypes: YES
Compiler for C supports arguments -Wredundant-decls: YES
Compiler for C supports arguments -Wshadow: YES
Found pkg-config: /usr/bin/pkg-config (1.9.4)
Run-time dependency yaml-0.1 found: YES 0.2.5
Run-time dependency valgrind found: YES 3.20.0
Program python3 found: YES (/usr/bin/python3)
Program swig found: YES (/usr/bin/swig)
Compiler for C supports link arguments -Wl,--version-script=/home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/libfdt/version.lds: YES
Program flex found: YES (/usr/bin/flex)
Program bison found: YES (/usr/bin/bison)
Check usable header "fnmatch.h" : YES
Program ../setup.py found: YES (/home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/pylibfdt/../setup.py)
Library dl found: YES
Program run_tests.sh found: YES (/home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/tests/run_tests.sh)
Build targets in project: 84

dtc 1.6.0

  User defined options
    auto_features : enabled
    bindir        : /usr/bin
    buildtype     : plain
    datadir       : /usr/share
    includedir    : /usr/include
    infodir       : /usr/share/info
    libdir        : /usr/lib64
    libexecdir    : /usr/libexec
    localedir     : /usr/share/locale
    localstatedir : /var
    mandir        : /usr/share/man
    prefix        : /usr
    sbindir       : /usr/sbin
    sharedstatedir: /var/lib
    sysconfdir    : /etc
    wrap_mode     : nodownload
    python        : enabled
    tools         : true
    valgrind      : enabled
    yaml          : enabled

Found ninja-1.11.1 at /usr/bin/ninja
+ /usr/bin/meson compile -C x86_64-redhat-linux-gnu -j 48 --verbose
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/x86_64-redhat-linux-gnu -j 48 -v
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/x86_64-redhat-linux-gnu'
[1/346] /usr/bin/flex -o dtc.p/dtc-lexer.l.lex.c ../dtc-lexer.l
[2/346] /usr/bin/flex -o convert-dtsv0.p/convert-dtsv0-lexer.l.lex.c ../convert-dtsv0-lexer.l

[..]

[100/346] /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/pylibfdt/../setup.py --quiet --top-builddir /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/x86_64-redhat-linux-gnu build_ext --build-lib=/home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/x86_64-redhat-linux-gnu/pylibfdt
/usr/lib/python3.8/site-packages/setuptools/__init__.py:85: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated. Requirements should be satisfied by a PEP 517 installer. If you are using pip, you can try `pip install --use-pep517`.
  dist.fetch_build_eggs(dist.setup_requires)
WARNING: The wheel package is not available.
```
</details>

And install fails with
<details>

```console
+ cd dtc-1.7.0
+ DESTDIR=/home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64
+ /usr/bin/meson install -C x86_64-redhat-linux-gnu --no-rebuild
Installing libfdt/libfdt.so.1.6.0 to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/lib64
Installing libfdt/libfdt.a to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/lib64
Installing convert-dtsv0 to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/bin
Installing dtc to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/bin
Installing fdtdump to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/bin
Installing fdtget to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/bin
Installing fdtput to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/bin
Installing fdtoverlay to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/bin
Installing /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/libfdt/fdt.h to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/include/
Installing /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/libfdt/libfdt.h to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/include/
Installing /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/libfdt/libfdt_env.h to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/include/
Installing /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/x86_64-redhat-linux-gnu/meson-private/libfdt.pc to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/lib64/pkgconfig
Installing /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/dtdiff to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/bin
Installing symlink pointing to libfdt.so.1.6.0 to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/lib64/libfdt.so.1
Installing symlink pointing to libfdt.so.1 to /home/tkloczko/rpmbuild/BUILDROOT/dtc-1.7.0-2.fc35.x86_64/usr/lib64/libfdt.so
Running custom install script '/home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/pylibfdt/../setup.py --quiet --top-builddir /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/x86_64-redhat-linux-gnu install --prefix=/usr --root=$DESTDIR'
/usr/lib/python3.8/site-packages/setuptools/__init__.py:85: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated. Requirements should be satisfied by a PEP 517 installer. If you are using pip, you can try `pip install --use-pep517`.
  dist.fetch_build_eggs(dist.setup_requires)
WARNING: The wheel package is not available.
Traceback (most recent call last):
  File "/home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/pylibfdt/../setup.py", line 50, in <module>
    setup(
  File "/usr/lib/python3.8/site-packages/setuptools/__init__.py", line 108, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 147, in setup
    _setup_distribution = dist = klass(attrs)
  File "/usr/lib/python3.8/site-packages/setuptools/dist.py", line 488, in __init__
    _Distribution.__init__(
  File "/usr/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 283, in __init__
    self.finalize_options()
  File "/usr/lib/python3.8/site-packages/setuptools/dist.py", line 912, in finalize_options
    ep(self)
  File "/usr/lib/python3.8/site-packages/setuptools/dist.py", line 932, in _finalize_setup_keywords
    ep.load()(self, ep.name, value)
  File "/usr/lib/python3.8/site-packages/setuptools_scm/integration.py", line 91, in version_keyword
    _assign_version(dist, config)
  File "/usr/lib/python3.8/site-packages/setuptools_scm/integration.py", line 63, in _assign_version
    _version_missing(config)
  File "/usr/lib/python3.8/site-packages/setuptools_scm/__init__.py", line 108, in _version_missing
    raise LookupError(
LookupError: setuptools-scm was unable to detect version for /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0.

Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.

For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
FAILED: install script '/home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/pylibfdt/../setup.py --quiet --top-builddir /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/x86_64-redhat-linux-gnu install --prefix=/usr --root=$DESTDIR' exit code 1, stopped
```
</details>

Typical pep517 build and istall procedure without pip consit from:
 - `python -sBm build -w --no-isolation`
    this will generate .wh in dist/ subdirectory
 - `python -sBm installer dist/*.whl --destdir $(DESTDIR)`

dgibson, 2023-04-13:
I've hit a bunch of problems with setuptools stuff as well.  Unfortunately, I don't really know enough about how it works to debug and fix it.

kloczek, 2023-04-13:
Please play with those two commands which I gave you.
I'm using tehm to build all my +1.1k rpm packages with pyton modules.

You need to install in buiuld env modules: wheel, builder, setuptools and installer (meson provides routine in its `python` module to check python modules dependencies).

dgibson, 2023-05-14:
I barely have time to keep up with the most minimal maintenance of dtc.  I'm afraid "playing with" some commands to figure out what's going on just isn't going to happen.  If you want this fixed, you're going to need to give me some muc more specific information about what needs to change here.

kloczek, 2023-10-30:
FYI: I found workaround by build dtc with disabled python support and than build python module using pep517 build procedure however I've hit issue with test suite which fails in two units.
<details>

```console
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/x86_64-redhat-linux-gnu'
ninja: no work to do.
1/1 run-test        FAIL              6.10s   exit status 1
>>> NO_PYTHON=1 MALLOC_PERTURB_=53 /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/tests/run_tests.sh
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
Listing only the last 100 lines from a long log.
dtbs_equal_ordered -n test_tree1.dtb test_tree1_wrong5.test.dtb:        PASS
dtbs_equal_ordered -n test_tree1.dtb test_tree1_wrong6.test.dtb:        PASS
dtbs_equal_ordered -n test_tree1.dtb test_tree1_wrong7.test.dtb:        PASS
dtbs_equal_ordered -n test_tree1.dtb test_tree1_wrong8.test.dtb:        PASS
dtbs_equal_ordered -n test_tree1.dtb test_tree1_wrong9.test.dtb:        PASS
dtbs_equal_unordered test_tree1.dtb test_tree1.dtb:     PASS
dtbs_equal_unordered test_tree1.dtb test_tree1.dtb.reversed.test.dtb:   PASS
dtbs_equal_unordered test_tree1.dtb.reversed.test.dtb test_tree1.dtb:   PASS
dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong1.test.dtb:      PASS
dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong2.test.dtb:      PASS
dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong3.test.dtb:      PASS
dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong4.test.dtb:      PASS
dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong5.test.dtb:      PASS
dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong6.test.dtb:      PASS
dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong7.test.dtb:      PASS
dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong8.test.dtb:      PASS
dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong9.test.dtb:      PASS
dtc -I dtb -O dtb -s -o test_tree1.dtb.sorted.test.dtb test_tree1.dtb:  PASS
dtbs_equal_unordered test_tree1.dtb test_tree1.dtb.sorted.test.dtb:     PASS
dtc -I dtb -O dtb -s -o test_tree1.dtb.reversed.sorted.test.dtb test_tree1.dtb.reversed.test.dtb:       PASS
dtbs_equal_unordered test_tree1.dtb.reversed.test.dtb test_tree1.dtb.reversed.sorted.test.dtb:  PASS
dtbs_equal_ordered test_tree1.dtb.sorted.test.dtb test_tree1.dtb.reversed.sorted.test.dtb:      PASS
dtc -O dtb -o label01.dts.fdtget.test.dtb /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/tests/label01.dts:    PASS
fdtget-runtest.sh "MyBoardName" label01.dts.fdtget.test.dtb / model:    PASS
fdtget-runtest.sh "MyBoardName MyBoardFamilyName" label01.dts.fdtget.test.dtb / compatible:     PASS
fdtget-runtest.sh "77 121 66 111 97 114 100 78 97 109 101 0 77 121 66 111 97 114 100 70 97 109 105 108 121 78 97 109 101 0" -t bu label01.dts.fdtget.test.dtb / compatible:     PASS
fdtget-runtest.sh "MyBoardName MyBoardFamilyName" -t s label01.dts.fdtget.test.dtb / compatible:        PASS
fdtget-runtest.sh "32768" label01.dts.fdtget.test.dtb /cpus/PowerPC,970@1 d-cache-size: PASS
fdtget-runtest.sh "8000" -tx label01.dts.fdtget.test.dtb /cpus/PowerPC,970@1 d-cache-size:      PASS
fdtget-runtest.sh "61 62 63 0" -tbx label01.dts.fdtget.test.dtb /randomnode tricky1:    PASS
fdtget-runtest.sh "a b c d de ea ad be ef" -tbx label01.dts.fdtget.test.dtb /randomnode blob:   PASS
fdtget-runtest.sh "MyBoardName\0MyBoardFamilyName\0" -tr label01.dts.fdtget.test.dtb / compatible:      PASS
fdtget-runtest.sh "\012\013\014\015\336\352\255\276\357" -tr label01.dts.fdtget.test.dtb /randomnode blob:      PASS
../fdtget -tlx label01.dts.fdtget.test.dtb /randomnode mixed {!= 0}:    PASS
fdtget-runtest.sh "6162 6300 1234 0 a 0 b 0 c" -thx label01.dts.fdtget.test.dtb /randomnode mixed:      PASS
fdtget-runtest.sh "61 62 63 0 12 34 0 0 0 a 0 0 0 b 0 0 0 c" -thhx label01.dts.fdtget.test.dtb /randomnode mixed:       PASS
../fdtget -ts label01.dts.fdtget.test.dtb /randomnode doctor-who {!= 0}:        PASS
fdtget-runtest.sh "MyBoardName\nmemory" -ts label01.dts.fdtget.test.dtb / model /memory device_type:    PASS
../fdtget -tx label01.dts.fdtget.test.dtb /randomnode doctor-who {!= 0}:        PASS
fdtget-runtest.sh "<the dead silence>" -tx -d <the dead silence> label01.dts.fdtget.test.dtb /randomnode doctor-who:    PASS
fdtget-runtest.sh "<blink>" -tx -d <blink> label01.dts.fdtget.test.dtb /memory doctor-who:      PASS
dtc -O dtb -p 2055 -o label01.dts.fdtput.test.dtb /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/tests/label01.dts:    PASS
fdtput-runtest.sh a_model label01.dts.fdtput.test.dtb / model -ts a_model:      PASS
fdtput-runtest.sh board1 board2 label01.dts.fdtput.test.dtb / compatible -ts board1 board2:     PASS
fdtput-runtest.sh board1 board2 label01.dts.fdtput.test.dtb / compatible -ts board1 board2:     PASS
fdtput-runtest.sh 32768 label01.dts.fdtput.test.dtb /cpus/PowerPC,970@1 d-cache-size  32768:    PASS
fdtput-runtest.sh 8001 label01.dts.fdtput.test.dtb /cpus/PowerPC,970@1 d-cache-size -tx 0x8001: PASS
fdtput-runtest.sh 2 3 12 label01.dts.fdtput.test.dtb /randomnode tricky1 -tbi 02 003 12:        PASS
fdtput-runtest.sh a b c ea ad be ef label01.dts.fdtput.test.dtb /randomnode blob -tbx a b c ea ad be ef:        PASS
[... cut ...]
*   Bad configuration:  1
* Strange test result:  0
**********
―――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

Summary of Failures:

1/1 run-test FAIL              6.10s   exit status 1

Ok:                 0
Expected Fail:      0
Fail:               1
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```
</details>

May I ask to have look on above output? 🤔

dgibson, 2023-11-04:
On Sun, Oct 29, 2023 at 06:17:59PM -0700, Tomasz Kłoczko wrote:
> FYI: I found workaround by build dtc with disabled python support and than build python module using pep517 build procedure however I've hit issue with test suite which fails in two units.
> <details>
> 
> ```console
> + /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
> ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/x86_64-redhat-linux-gnu'
> ninja: no work to do.
> 1/1 run-test        FAIL              6.10s   exit status 1
> >>> NO_PYTHON=1 MALLOC_PERTURB_=53 /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/tests/run_tests.sh
> ――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
> Listing only the last 100 lines from a long log.
> dtbs_equal_ordered -n test_tree1.dtb test_tree1_wrong5.test.dtb:        PASS
> dtbs_equal_ordered -n test_tree1.dtb test_tree1_wrong6.test.dtb:        PASS
> dtbs_equal_ordered -n test_tree1.dtb test_tree1_wrong7.test.dtb:        PASS
> dtbs_equal_ordered -n test_tree1.dtb test_tree1_wrong8.test.dtb:        PASS
> dtbs_equal_ordered -n test_tree1.dtb test_tree1_wrong9.test.dtb:        PASS
> dtbs_equal_unordered test_tree1.dtb test_tree1.dtb:     PASS
> dtbs_equal_unordered test_tree1.dtb test_tree1.dtb.reversed.test.dtb:   PASS
> dtbs_equal_unordered test_tree1.dtb.reversed.test.dtb test_tree1.dtb:   PASS
> dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong1.test.dtb:      PASS
> dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong2.test.dtb:      PASS
> dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong3.test.dtb:      PASS
> dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong4.test.dtb:      PASS
> dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong5.test.dtb:      PASS
> dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong6.test.dtb:      PASS
> dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong7.test.dtb:      PASS
> dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong8.test.dtb:      PASS
> dtbs_equal_unordered -n test_tree1.dtb test_tree1_wrong9.test.dtb:      PASS
> dtc -I dtb -O dtb -s -o test_tree1.dtb.sorted.test.dtb test_tree1.dtb:  PASS
> dtbs_equal_unordered test_tree1.dtb test_tree1.dtb.sorted.test.dtb:     PASS
> dtc -I dtb -O dtb -s -o test_tree1.dtb.reversed.sorted.test.dtb test_tree1.dtb.reversed.test.dtb:       PASS
> dtbs_equal_unordered test_tree1.dtb.reversed.test.dtb test_tree1.dtb.reversed.sorted.test.dtb:  PASS
> dtbs_equal_ordered test_tree1.dtb.sorted.test.dtb test_tree1.dtb.reversed.sorted.test.dtb:      PASS
> dtc -O dtb -o label01.dts.fdtget.test.dtb /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/tests/label01.dts:    PASS
> fdtget-runtest.sh "MyBoardName" label01.dts.fdtget.test.dtb / model:    PASS
> fdtget-runtest.sh "MyBoardName MyBoardFamilyName" label01.dts.fdtget.test.dtb / compatible:     PASS
> fdtget-runtest.sh "77 121 66 111 97 114 100 78 97 109 101 0 77 121 66 111 97 114 100 70 97 109 105 108 121 78 97 109 101 0" -t bu label01.dts.fdtget.test.dtb / compatible:     PASS
> fdtget-runtest.sh "MyBoardName MyBoardFamilyName" -t s label01.dts.fdtget.test.dtb / compatible:        PASS
> fdtget-runtest.sh "32768" label01.dts.fdtget.test.dtb ***@***.*** d-cache-size: PASS
> fdtget-runtest.sh "8000" -tx label01.dts.fdtget.test.dtb ***@***.*** d-cache-size:      PASS
> fdtget-runtest.sh "61 62 63 0" -tbx label01.dts.fdtget.test.dtb /randomnode tricky1:    PASS
> fdtget-runtest.sh "a b c d de ea ad be ef" -tbx label01.dts.fdtget.test.dtb /randomnode blob:   PASS
> fdtget-runtest.sh "MyBoardName\0MyBoardFamilyName\0" -tr label01.dts.fdtget.test.dtb / compatible:      PASS
> fdtget-runtest.sh "\012\013\014\015\336\352\255\276\357" -tr label01.dts.fdtget.test.dtb /randomnode blob:      PASS
> ../fdtget -tlx label01.dts.fdtget.test.dtb /randomnode mixed {!= 0}:    PASS
> fdtget-runtest.sh "6162 6300 1234 0 a 0 b 0 c" -thx label01.dts.fdtget.test.dtb /randomnode mixed:      PASS
> fdtget-runtest.sh "61 62 63 0 12 34 0 0 0 a 0 0 0 b 0 0 0 c" -thhx label01.dts.fdtget.test.dtb /randomnode mixed:       PASS
> ../fdtget -ts label01.dts.fdtget.test.dtb /randomnode doctor-who {!= 0}:        PASS
> fdtget-runtest.sh "MyBoardName\nmemory" -ts label01.dts.fdtget.test.dtb / model /memory device_type:    PASS
> ../fdtget -tx label01.dts.fdtget.test.dtb /randomnode doctor-who {!= 0}:        PASS
> fdtget-runtest.sh "<the dead silence>" -tx -d <the dead silence> label01.dts.fdtget.test.dtb /randomnode doctor-who:    PASS
> fdtget-runtest.sh "<blink>" -tx -d <blink> label01.dts.fdtget.test.dtb /memory doctor-who:      PASS
> dtc -O dtb -p 2055 -o label01.dts.fdtput.test.dtb /home/tkloczko/rpmbuild/BUILD/dtc-1.7.0/tests/label01.dts:    PASS
> fdtput-runtest.sh a_model label01.dts.fdtput.test.dtb / model -ts a_model:      PASS
> fdtput-runtest.sh board1 board2 label01.dts.fdtput.test.dtb / compatible -ts board1 board2:     PASS
> fdtput-runtest.sh board1 board2 label01.dts.fdtput.test.dtb / compatible -ts board1 board2:     PASS
> fdtput-runtest.sh 32768 label01.dts.fdtput.test.dtb ***@***.*** d-cache-size  32768:    PASS
> fdtput-runtest.sh 8001 label01.dts.fdtput.test.dtb ***@***.*** d-cache-size -tx 0x8001: PASS
> fdtput-runtest.sh 2 3 12 label01.dts.fdtput.test.dtb /randomnode tricky1 -tbi 02 003 12:        PASS
[... cut ...]
> Fail:               1
> Unexpected Pass:    0
> Skipped:            0
> Timeout:            0
> ```
> </details>
> 
> May I ask to have look on above output? 🤔 

Unfortunately, while it's showing the overall failure, I'm not seeing
any of the individual failing cases.  So there's really no information
for me to go on here.

-- 
David Gibson			| I'll have my music baroque, and my code
david AT gibson.dropbear.id.au	| minimalist, thank you.  NOT _the_ _other_
				| _way_ _around_!
http://www.ozlabs.org/~dgibson

kloczek, 2023-11-04:
> Unfortunately, while it's showing the overall failure, I'm not seeing any of the individual failing cases. So there's really no information for me to go on here.

Any hint what I can try to do to diagnose this issue? 🤔

dgibson, 2023-11-04:
Hmm.   Does the failure also occur if you build with `make` instead of `meson`?

hybotics, 2023-12-27:
I am also having problems relating to the setuptools version on my system. I run Ubuntu MATE' 22.04 on a Ryzen 9 laptop.
```$ cd dtc/
hybotics@shuttle
~/Repos/Github/dgibson/dtc
$ make
	 DEP tests/dumptrees.c
	 DEP tests/trees.S
	 DEP tests/testutils.c
	 DEP tests/value-labels.c
	 DEP tests/asm_tree_dump.c
	 DEP tests/truncated_memrsv.c
	 DEP tests/truncated_string.c
	 DEP tests/truncated_property.c
	 DEP tests/fs_tree1.c
	 DEP tests/check_full.c
	 DEP tests/check_header.c
	 DEP tests/check_path.c
	 DEP tests/overlay_bad_fixup.c
	 DEP tests/overlay.c
	 DEP tests/subnode_iterate.c
	 DEP tests/property_iterate.c
	 DEP tests/integer-expressions.c
	 DEP tests/utilfdt_test.c
	 DEP tests/path_offset_aliases.c
	 DEP tests/add_subnode_with_nops.c
	 DEP tests/dtbs_equal_unordered.c
	 DEP tests/dtb_reverse.c
	 DEP tests/dtbs_equal_ordered.c
	 DEP tests/extra-terminating-null.c
	 DEP tests/relref_merge.c
	 DEP tests/incbin.c
	 DEP tests/boot-cpuid.c
	 DEP tests/phandle_format.c
	 DEP tests/path-references.c
	 DEP tests/references.c
	 DEP tests/string_escapes.c
	 DEP tests/propname_escapes.c
	 DEP tests/appendprop2.c
	 DEP tests/appendprop1.c
	 DEP tests/del_node.c
	 DEP tests/del_property.c
	 DEP tests/setprop.c
	 DEP tests/set_name.c
	 DEP tests/rw_oom.c
	 DEP tests/rw_tree1.c
	 DEP tests/open_pack.c
	 DEP tests/nopulate.c
	 DEP tests/mangle-layout.c
	 DEP tests/move_and_save.c
	 DEP tests/sw_states.c
	 DEP tests/sw_tree1.c
	 DEP tests/nop_node.c
	 DEP tests/nop_property.c
	 DEP tests/setprop_inplace.c
	 DEP tests/stringlist.c
	 DEP tests/appendprop_addrrange.c
	 DEP tests/addr_size_cells2.c
	 DEP tests/addr_size_cells.c
	 DEP tests/notfound.c
	 DEP tests/sized_cells.c
	 DEP tests/char_literal.c
	 DEP tests/get_next_tag_invalid_prop_len.c
	 DEP tests/get_alias.c
	 DEP tests/node_offset_by_compatible.c
	 DEP tests/node_check_compatible.c
	 DEP tests/node_offset_by_phandle.c
	 DEP tests/node_offset_by_prop_value.c
	 DEP tests/parent_offset.c
	 DEP tests/supernode_atdepth_offset.c
	 DEP tests/get_path.c
	 DEP tests/get_phandle.c
	 DEP tests/get_prop_offset.c
	 DEP tests/getprop.c
	 DEP tests/get_name.c
	 DEP tests/path_offset.c
	 DEP tests/subnode_offset.c
	 DEP tests/find_property.c
	 DEP tests/root_node.c
	 DEP tests/get_mem_rsv.c
	 DEP libfdt/fdt_check.c
	 DEP libfdt/fdt_overlay.c
	 DEP libfdt/fdt_addresses.c
	 DEP libfdt/fdt_empty_tree.c
	 DEP libfdt/fdt_strerror.c
	 DEP libfdt/fdt_rw.c
	 DEP libfdt/fdt_sw.c
	 DEP libfdt/fdt_wip.c
	 DEP libfdt/fdt_ro.c
	 DEP libfdt/fdt.c
	 DEP util.c
	 DEP fdtoverlay.c
	 DEP fdtput.c
	 DEP fdtget.c
	 DEP fdtdump.c
	 LEX convert-dtsv0-lexer.lex.c
	 DEP convert-dtsv0-lexer.lex.c
	 DEP srcpos.c
	 BISON dtc-parser.tab.c
	 DEP dtc-parser.tab.c
	 LEX dtc-lexer.lex.c
	 DEP dtc-lexer.lex.c
	 DEP treesource.c
	 DEP livetree.c
	 DEP fstree.c
	 DEP flattree.c
	 DEP dtc.c
	 DEP data.c
	 DEP checks.c
	CHK version_gen.h
	UPD version_gen.h
	 DEP util.c
	CHK version_gen.h
	 CC srcpos.o
	 CC util.o
	 CC convert-dtsv0-lexer.lex.o
	 LD convert-dtsv0
	 CC dtc.o
	 CC checks.o
	 CC data.o
	 CC flattree.o
	 CC fstree.o
	 CC livetree.o
	 CC treesource.o
	 CC dtc-lexer.lex.o
	 CC dtc-parser.tab.o
	 LD dtc
	 CC fdtdump.o
	 LD fdtdump
	 CC fdtget.o
	 CC libfdt/fdt.o
	 CC libfdt/fdt_ro.o
	 CC libfdt/fdt_wip.o
	 CC libfdt/fdt_sw.o
	 CC libfdt/fdt_rw.o
	 CC libfdt/fdt_strerror.o
	 CC libfdt/fdt_empty_tree.o
	 CC libfdt/fdt_addresses.o
	 CC libfdt/fdt_overlay.o
	 CC libfdt/fdt_check.o
	 LD libfdt/libfdt.so.1.7.0
	 LD fdtget
	 CC fdtput.o
	 LD fdtput
	 CC fdtoverlay.o
	 LD fdtoverlay
	 AR libfdt/libfdt.a
make[1]: Entering directory '/home/hybotics/Repos/Github/dgibson/dtc'
	CHK version_gen.h
	 PYMOD pylibfdt/_libfdt.so
/home/hybotics/VirtualEnv/zephyr/lib/python3.10/site-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
  warnings.warn(
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
/home/hybotics/Repos/Github/dgibson/dtc/.eggs/setuptools_scm-8.0.4-py3.10.egg/setuptools_scm/_integration/setuptools.py:30: RuntimeWarning: 
ERROR: setuptools==59.6.0 is used in combination with setuptools_scm>=8.x

Your build configuration is incomplete and previously worked by accident!
setuptools_scm requires setuptools>=61

Suggested workaround if applicable:
 - migrating from the deprecated setup_requires mechanism to pep517/518
   and using a pyproject.toml to declare build dependencies
   which are reliably pre-installed before running the build tools

  warnings.warn(
WARNING setuptools_scm.pyproject_reading toml section missing 'pyproject.toml does not contain a tool.setuptools_scm section'
make[1]: Leaving directory '/home/hybotics/Repos/Github/dgibson/dtc'```

dgibson, 2024-01-25:
@kloczek sorry, no progress.  This is deep into the details of how setuptoools / distutils etc. work.  I'm not familiar with it and I don't really have time to learn.  @sjg20 any chance you can tackle this?

sjg20, 2026-11-05:
This should be fixed now. The setuptools-scm error was fixed by:

cd3e230 pylibfdt: use fallback version in tarballs

which was in v1.7.1

Since v1.8.0 the Python bindings build natively with meson and meson-python, so setup.py is gone.

I checked ‘meson install’ with -Dpython=enabled from an export with no git metadata. It fails on v1.7.0 as reported, but works on v1.7.1

The test failures look like a bug in the test script. Running v1.7.0 with -Dpython=disabled gives the same counts (2 FAIL, 1 Bad configuration), from the cell-overflow tests using a relative path which is not found when running under meson. That was fixed by this patch, also in v1.7.1

32174a6 meson: Fix cell overflow tests when running from meson

Please reopen this if you see a problem with v1.8.1

closed 2026-11-05 by sjg20