dgibson/dtc #106: Multiple-definition of fdt target in Meson definitions

ernestask, 2023-07-27:
When configuring the project with Meson, this is the result:
```
Tried to create target "fdt", but a target of that name already exists.
```

https://www.spinics.net/lists/devicetree-compiler/msg03863.html would have fixed it, but it was never accepted. The concern was that removing the duplicate target would somehow make Meson not spit out targets for both the shared and static library, but, as Marc-André explained, this should be set through the `default_library` option. `both` is a valid setting.

ernestask, 2023-07-27:
Moreover, there should probably be an Actions workflow to perform a test-build with Meson.

ernestask, 2023-07-27:
https://mesonbuild.com/Builtin-options.html#core-options

dgibson, 2023-07-28:
Ugh.  The unfortunate truth is that the meson stuff was applied by someone else, I don't really have the knowledge to maintain it, nor the time to learn what I need.

@elmarco any chance you could look into this?

dgibson, 2023-07-28:
Also, it would be good to have github actions for lots of things, including this.  Learning how to write and use github actions is another thing I really haven't had time for.

ernestask, 2023-07-28:
Since I’m currently toying with an old project that uses libfdt, I could look into implementing at least some of those wishlist items.

dgibson, 2024-01-29:
I think this was fixed by [this commit](https://github.com/dgibson/dtc/commit/3fbfdd08afd2a7a25b27433f6f5678c0fe694721).  @ernestask, could you verify?

sjg20, 2026-11-05:
Yes this is fixed now. The error came from defining the library twice, fixed by these two commits, both in v1.7.1:

3fbfdd0 libfdt: fix duplicate meson target
da39ee0 libfdt: rework shared/static libraries

The first fixes default_library=static and the second fixes default_library=both - I can reproduce the error on v1.7.0 with both of those settings, and it is gone in v1.7.1 and later.

Since v1.8.0, default_library is also respected, so ‘shared’ and ‘static’ build only the library requested. BTW CI now has Meson jobs as well, in .github/workflows/build.yml

Please reopen this if you still see a problem.

closed 2026-11-05 by sjg20