dgibson/dtc #119: fdt.c crashed at t absoffset = offset + fdt_off_dt_struct(fdt);

symdeb, 2023-09-10:
I am using the ECO system 5.0 TF-A build environment from ST and built the -tfa-.t When booting the board it crashed without a message..I put tons of NOTICE messages in the code. Finally nailed it it down, It runs for 45 minutes before is stops in fdt.c.  Repeated this 8 times, the point it stops is always the same (after DEBUG: fdt_offset_ptr 0); Aapproached ST but so far no clues. 

How do I identify which version of fdt.c ST is using ?
Has this been or is a know issues , if yes how to workaround it ?

Function: fdt_offset_ptr()

// PATCH: Add for debug 
#include <common/debug.h>

const void *fdt_offset_ptr(const void *fdt, int offset, unsigned int len)
{
	unsigned int uoffset = offset;

	NOTICE ("DEBUG: fdt_offset_ptr 0\n");

	unsigned int absoffset = offset + fdt_off_dt_struct(fdt);

	NOTICE ("DEBUG: fdt_offset_ptr 1\n");

dgibson, 2023-09-12:
> [STM32MP1 TF-A crash.zip](https://github.com/dgibson/dtc/files/12581274/STM32MP1.TF-A.crash.zip) I am using the ECO system 5.0 TF-A build environment from ST used for the STM32MP1. built the t-fa. 

From context, I'm assuming the STM32MP1 is a specific board, and ST are its makers.  I know nothing else specific about this.  I'm not sure what a "TF-A" means either.

> When booting the board from SD card (or via USB using the Cubeprogrammer) TF-A crashed without a message.. Put tons of NOTICE messages in the code. Finally nailed it it down, It runs for 45 minutes before it stops in fdt.c. Repeated this 12 times, the point it stops is always the same (after DEBUG: fdt_offset_ptr 0); it always happen at the 3rd node for power (vin, vdd and then the 3rd) of the DT even the node is replaced with another one. Approached ST for direct customer support, but so far no answers/clues.

That's  really weird.  As you can see that line is just an addition, which isn't an instruction which can cause a crash on most CPUs.  It does load a value `fdt_off_dt_struct(fdt)` from the fdt header, which is a bit more likely to cause problems.  However it sounds like it's not typically crashing on the *first* time here, so I'm not sure why a later call would fail.

Nonetheless, it's probably worth trying to add a temporary variable and load `fdt_off_dt_struct(fdt)` into it, then add a `NOTICE` then do the addition, to confirm which operation is causing the crash.  Seeing the actual values of `fdt` and `offset` might help too.

> How do I identify which version of fdt.c ST is using ? ST removed the version file.

If they've removed the version, I don't think there's any easy way to do this.  The only approach I can think of is just to compare it to various versions of the upstream source and try to find a match.

> Has this been, or is this a known issue, if yes how to workaround it ? What could be causes. Anything else to try.

No, certainly not a known issue - as noted, it's very odd behaviour.  In fact I suspect the bug is not actually here, but something before the call which crashes corrupting either the fdt pointer or something in the fdt itself.

symdeb, 2023-09-12:
Thanks for the reply  . It's indeed a very weird issue but there is nothing I can actually change. the whole  toolchain and source code is as provided by ST and the FDT is part it.  TF-A is the name  used for the FSBL
https://trustedfirmware-a.readthedocs.io/en/latest/index.html. 
is there a way to compile the DT and  decompile the DT with another standalone tool  that uses the same FDT library  to see it's reproduceable  using the attched DT ?

dgibson, 2023-09-12:
> Thanks for the reply . It's indeed a very weird issue but there is nothing I can actually change.

Uh... I'm slightly confused.  If you were able to add the `NOTICE` commands you should also be able to rearrange the load/addition there, yes?  What am I missing?

> the whole toolchain and source code is as provided by ST and the FDT is part it. TF-A is the name used for the FSBL https://trustedfirmware-a.readthedocs.io/en/latest/index.html. is there a way to [...]

What form did the dt itself come in?  Source (dts) or binary (dtb)?  Or is it built into the firmware image somehow?

You can use dtc (`git://git.kernel.org/pub/scm/utils/dtc/dtc.git`) to compile and decompile device trees.  It's not strictly speaking based on libfdt, but dtc and libfdt are part of the same upstream source tree, and they are cross-tested. 

>  I also wonder if there was an invalid pointer and wanted to add a check, but what are correct and incorrect values ?

There isn't a simple answer to that - it depends on the memory map of the firmware.  `0` or something unaligned is probably bad, but not necessarily, and those certainly aren't the only possibilities for a bad address.  The value changing between one call to `fdt_offset_ptr()` and the next would also be a bad sign.

symdeb, 2023-09-26:
the dts is in source form  I could not fix this, after power in only this is shown:

NOTICE:  CPU: STM32MP157AAA Rev.B
NOTICE:  Model: STMicroelectronics STM32MP157A-DK1 Discovery Board

The original source files used a PMIC I2C4  entry and inside that the regulator power entries . So I had a trial and error  thought  ... "why not move the additional entries for discrete regulator power to a 2nd root entry ". That  worked  and I could add as many nodes as I needed.  fdt code does not crash anymore.  But I wonder if the ST code will read/use the "2nd" root entry of the DT.
P.S: This issue seems not to happen with the DT's from u-Boot and OP-TEEm only TF-A. (BL2)

/ {
      memory:
     vdd_core:
     vdd:
     /// memory and  2 (regulator) entries here , adding one more stops the boot
};

.. other entries for &bsec, &clk_hse, &cpu0m, &cpu1

// instead of the I2C entry with PMIC regulator nodes,  add another root node for regulators 
/{
  vdd_ddr: 
  vtt_ddr:
  v3v3:
  vdd_usb:
  etc..
};

dgibson, 2023-10-08:
Sorry, I can barely follow what you're talking about here.  It really seems like this is an issue with the broader board support package rather than in libfdt itself.

symdeb, 2023-10-09:
Sorry for that. its a hard failure,  100% reproduce-able with the ST ECO source package. Perhaps it is ST unique. Anyhow as long as there are not more than two regulator  nodes in the TF-A root (for example vin and vdd and remove v3v3 as the example above ), the freeze does not happen. When I have time in the future will look at this further,

closed 2023-10-09 by symdeb