dgibson/dtc #120: how can I download the archive inside debootstrap environment

d2nikolic, 2023-09-19:
I see no clear link to use curl and download dtc-main archive. Only a link for .zip download, but debootstrap has no 'unzip' command in it. Any chance you can tell me how to pull tgz using curl?
Thank you,
Dragan

dgibson, 2023-09-20:
I have no idea what archive you're looking at that has .zip downloads.  The primary download archive for dtc is at [https://mirrors.edge.kernel.org/pub/software/utils/dtc/](https://mirrors.edge.kernel.org/pub/software/utils/dtc/) which is all .tar.gz.

d2nikolic, 2023-09-20:
What used to work in bullseye Debian 11.7 is the following:

curl -kL https://github.com/RedPitaya/dtc/archive/overlays.tar.gz -o dtc.tar.gz
tar zxvf dtc.tar.gz
cd dtc-overlays
# Fix duplicate declaration of yylloc
sed -i -e '/YYLTYPE yylloc/d' dtc-lexer.l
make
make install PREFIX=/usr

but now in Debian bookworm this fails

dgibson, 2023-09-20:
That's a third party fork of dtc by someone else ("Red Pitaya", whoever they are), itself based on another fork of dtc.  It's not my archive, and it's a modified version of my code, so I really can't answer anything about it.  You'll need to contact Red Pitaya.

closed 2023-09-20 by dgibson