dgibson/dtc #134: python extension build with meson is always quiet

sharkcz, 2024-05-29:
Building the python extension with `meson` is always quiet as it unconditionally passes the `--quiet` option while the Makefile build only sets the `--quiet` options when the build is not verbose (`V` is not set). It makes it impossible to verify the options passed to the compiler and linker for example.

blmaier, 2024-05-29:
This could be supported by adding a custom meson_options.txt e.g. `option('setuptools-verbose')`.

Preferably it would be automatic with `meson build --verbose`. But the compilation is being done indirectly by Python setuptools so Meson's backend doesn't know about it.

sharkcz, 2024-05-29:
yes, it would nice if the `--verbose` option was reflected somewhere in the setup and could be used in the build script

blmaier, 2025-06-12:
Building with `meson compile -v` this now prints out the compile commands for the python library, so this should be fixed.

For example this is what I see when compiling.
```
[18/337] ccache cc -Ipylibfdt/_libfdt.cpython-313-x86_64-linux-gnu.so.p -Ipylibfdt -I../pylibfdt -Ilibfdt -I../libfdt -I/usr/include/python3.13 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -O0 -g -Wpointer-arith -Wcast-qual -Wnested-externs -Wstrict-prototypes -Wmissing-prototypes -Wredundant-decls -Wshadow -Wsuggest-attribute=format -Wwrite-strings -DFDT_ASSUME_MASK=0 -DNO_YAML -DNO_VALGRIND -fPIC -DPY_SSIZE_T_CLEAN -Wno-cast-qual -Wno-missing-prototypes -Wno-redundant-decls -MD -MQ pylibfdt/_libfdt.cpython-313-x86_64-linux-gnu.so.p/meson-generated_.._libfdt.c.o -MF pylibfdt/_libfdt.cpython-313-x86_64-linux-gnu.so.p/meson-generated_.._libfdt.c.o.d -o pylibfdt/_libfdt.cpython-313-x86_64-linux-gnu.so.p/meson-generated_.._libfdt.c.o -c pylibfdt/libfdt.c
...
[215/337] cc  -o pylibfdt/_libfdt.cpython-313-x86_64-linux-gnu.so pylibfdt/_libfdt.cpython-313-x86_64-linux-gnu.so.p/meson-generated_.._libfdt.c.o -Wl,--as-needed -Wl,--allow-shlib-undefined -shared -fPIC '-Wl,-rpath,$ORIGIN/../libfdt' -Wl,-rpath-link,/local_build_dir/blmaier/dtc-134-meson-python-quiet/dtc/build/libfdt libfdt/libfdt.so.1.7.2
```

sharkcz, 2025-06-20:
looks good here as well