dgibson/dtc #162: Disable asserts if force flag enabled

bmx666, 2024-12-18:
I have some dtb to analyze from one of embedded device, it contains some errors, but during conversation from dtb to dts tool throws assert.

```
dtc -f -I dtb -O dts broken.dtb > broken.dts
```

```
dtc: livetree.c:437: propval_cell: Assertion `prop->val.len == sizeof(cell_t)' failed.
```

Could you please ignore assert if "force" has been enabled?
```
cell_t propval_cell(struct property *prop)
{
        if (!force)
                assert(prop->val.len == sizeof(cell_t));
        return fdt32_to_cpu(*((fdt32_t *)prop->val.val));
}

cell_t propval_cell_n(struct property *prop, unsigned int n)
{
        if (!force)
                assert(prop->val.len / sizeof(cell_t) > n);
        return fdt32_to_cpu(*((fdt32_t *)prop->val.val + n));
}
```

dgibson, 2024-12-24:
@bmx666 that's certainly a bug, but the proposed solution isn't quite right.  The problem lies in the caller of `propval_cell()` - it shouldn't be calling that on untrusted input without first verifying that the property has the expected form.

Can you attach the `dtb` causing the problem, then I can work out exactly what path is causing this and fix it.

bmx666, 2024-12-24:
[Spreadtrum_SC7731e.dtb.zip](https://github.com/user-attachments/files/18242047/Spreadtrum_SC7731e.dtb.zip)

dgibson, 2024-12-25:
I've had a preliminary look and it sems like the problem is in the graph checks.  As an interim approach, you should be able to disable those with `-Wno-graph_nodes`.

bmx666, 2024-12-25:
Probably I'm not the first and the last. The most people use dtc from package delivery system in OS. Because I had experience early it take me 5 min to find and fix issue to generate at least some dts. My question, do you have plans to fix it or just left it as-is? If you don't want to fix it, just close the current issue.

dgibson, 2024-12-26:
Sorry, I wasn't clear.  Yes I plan to fix this, probably soon - it doesn't look too hard.  But, it will be at least a few days, probably not until the new year.  In the meantime you can disable the `graph_nodes` check as a workaround.

sjg20, 2026-11-05:
Confirmed that this problem is still present