dgibson/dtc #163: Test failure with Glibc-2.41 because the tests/*.S lack non-executable stack marks

xry111, 2025-02-04:
Those .S files lack the mark for non-executable stack:

```
.section    .note.GNU-stack,"",@progbits
```

so the shared libraries in the test suite will be marked as "requiring executable stack" if using one of them, for example:

```
$ readelf -a tests/oasm_aliases.dts.test.so | grep GNU_STACK -A1
  GNU_STACK      0x0000000000000000 0x0000000000000000 0x0000000000000000
                 0x0000000000000000 0x0000000000000000  RWE    0x10
```

And since the 2.41 release, Glibc no longer allows `dlopen`ing such a shared library, quoting from its NEWS file:

```
* dlopen and dlmopen no longer make the stack executable if a shared
  library requires it, either implicitly because of a missing GNU_STACK
  ELF header (and default ABI permission having the executable bit set) 
  or explicitly because of the executable bit in GNU_STACK, and the
  stack is not already executable.  Instead, loading such objects will 
  fail.
```

So some tests fail because failure to `dlopen` the shared libraries.

dgibson, 2025-02-05:
Thanks for the report.  I don't have an easy way to test against glibc-2.41 right now, but I've made a draft patch which I think should fix it.

Could you please test the branch at https://github.com/dgibson/dtc/tree/bug163?  If that fixes the error for you, I'll merge it into mainline.

xry111, 2025-02-05:
Hmm it seems not enough...  The command

```
$CC -shared -o $1.test.so "$SRCDIR/data.S" $1.test.s
```

is building data.S and the .test.s file into two .o files, but the linker will mark the output as "requiring executable stack" if any of the .o files "requires" executable stack.

dgibson, 2025-02-05:
> Hmm it seems not enough... The command
> 
> ```
> $CC -shared -o $1.test.so "$SRCDIR/data.S" $1.test.s
> ```
> 
> is building data.S and the .test.s file into two .o files and then link them, but the linker will mark the output as "requiring executable stack" if any of the .o files "requires" executable stack.

Ah, I see.  Drat.

> diff --git a/flattree.c b/flattree.c
> index 1bcd808..f1f2294 100644
> --- a/flattree.c
> +++ b/flattree.c
> @@ -559,6 +559,7 @@ void dt_to_asm(FILE *f, struct dt_info *dti, int version)
>  	if (alignsize > 0)
>  		asm_emit_align(f, alignsize);
>  	emit_label(f, symprefix, "blob_abs_end");
> +	fprintf(f, "\t.section\t.note.GNU-stack,\"\",@progbits\n");
>  
>  	data_free(strbuf);
>  }
> 
> This works for me combined with your patch.

Right.  I thought about that, but I don't think it's a good idea.  It will work if the output from `-Oasm` is being used verbatim as an entire object, but one likely way to use it is to include it into another assembly file.  That might be for a bootloader which is likely to have its own linking requirements different from glibc's userspace conventions.  Let me try another approach.

dgibson, 2025-02-05:
I have another attempt at https://github.com/dgibson/dtc/tree/bug163.  Please give it a try.

xry111, 2025-02-05:
Yes it should work.  I had already used `CC="gcc -Wl,-z,noexecstack"` for work it around locally.

dgibson, 2025-02-05:
> Yes it should work. I had already used `CC="gcc -Wl,-z,noexecstack"` for work it around locally.

Ok.  But have you actually tested the new patch?  I'd like a confirmed report before merging it.

closed 2025-02-05 by dgibson

dgibson, 2025-02-05:
Fix pushed.  Thanks for the report.